siemens/kas #5: Incompatible with Python3.7

hANSIc99, 2018-11-02:
When I try to execute kas with python3.7, the following error message appears:

```
Traceback (most recent call last):
  File "/usr/local/bin/kas", line 7, in <module>
    from kas.kas import main
  File "/usr/local/lib/python3.7/site-packages/kas/kas.py", line 51, in <module>
    from .libkas import kasplugin
  File "/usr/local/lib/python3.7/site-packages/kas/libkas.py", line 168
    task = asyncio.async(repo.fetch_async())
                       ^
SyntaxError: invalid syntax
```

However, when I execute kas with python3.5 it runs without problems:
_**python3.5 -m kas**_

jan-kiszka, 2018-11-04:
Does this help? I don't have 3.7 around to test.

```
diff --git a/kas/libkas.py b/kas/libkas.py
index 2706796..2cf74f5 100644
--- a/kas/libkas.py
+++ b/kas/libkas.py
@@ -157,18 +157,22 @@ def find_program(paths, name):
     return None
 
 
+def _create_task(routine):
+    try:
+        creation_func = asyncio.ensure_future
+    except AttributeError:
+        # for Python < 3.5, avoiding the keyword 'async' introduced in 3.7
+        creation_func = getattr(asyncio, 'async')
+
+    return creation_func(routine)
+
 def repos_fetch(repos):
     """
         Fetches the list of repositories to the kas_work_dir.
     """
     tasks = []
     for repo in repos:
-        if not hasattr(asyncio, 'ensure_future'):
-            # pylint: disable=no-member,deprecated-method
-            task = asyncio.async(repo.fetch_async())
-        else:
-            task = asyncio.ensure_future(repo.fetch_async())
-        tasks.append(task)
+        tasks.append(_create_task(repo.fetch_async()))
 
     loop = asyncio.get_event_loop()
     loop.run_until_complete(asyncio.wait(tasks))
@@ -184,12 +188,7 @@ def repos_apply_patches(repos):
     """
     tasks = []
     for repo in repos:
-        if not hasattr(asyncio, 'ensure_future'):
-            # pylint: disable=no-member,deprecated-method
-            task = asyncio.async(repo.apply_patches_async())
-        else:
-            task = asyncio.ensure_future(repo.apply_patches_async())
-        tasks.append(task)
+        tasks.append(_create_task(repo.apply_patches_async()))
 
     loop = asyncio.get_event_loop()
     loop.run_until_complete(asyncio.wait(tasks))
```

But please use the mailing list for follow-ups. We consolidate all communication there, and I'll also propose this patch properly only on that channel.

jan-kiszka, 2018-11-05:
Patch sent, https://groups.google.com/forum/#!topic/kas-devel/YO17JUxCBQg, we still need feedback regarding 3.7, though.

closed 2019-03-19 by igaw