siemens/kas #7: multiconfig syntax is not documentaed

moto-timo, 2019-05-24:
Please let us know how to use the `multiconfig:*` feature. Do we put conf/multiconfig/foo.conf in local layer (meta-custom)?

moto-timo, 2019-05-24:
The syntax is:

```
target:
  - <target>
  - multiconfig:<multiconfig_name>:<target>
```

Where you would have ``conf/multiconfig/<multiconfig_name>.conf`` in your local layer (meta-custom).

For example:

```
target:
  - core-image-base
  - multiconfig:foo:core-image-minimal
```

For more details on multiconfig, see the [Yocto Project documentation](https://www.yoctoproject.org/docs/current/mega-manual/mega-manual.html#dev-building-images-for-multiple-targets-using-multiple-configurations).

jan-kiszka, 2019-05-24:
Yes, that multiconfig "just works" is not explicitly documented, only indirectly via changelog and format history. Feel free to propose something for the user doc. Just note that we have email-based patch workflow via the mailing list.

closed 2019-08-20 by jan-kiszka

vivien, 2021-03-22:
This means that _BBMULTICONFIG_ won't be set if _target_ doesn't contain an `mc:` prefix. This can be problematic if the recipe being built contains `do_task[mcdepends] = "mc:foo:bar:image:task"`.

Defining _BBMULTICONFIG_ manually in _local_conf_header_ isn't a solution because kas wouldn't have copied the multiconfig files to the build directory.

Am I correct?

jan-kiszka, 2021-03-22:
Just realized the duplication to #30... Please comment at least on the active issue or - even better in this case - open a new one for a new issues.

To answer your question in advance: yes, this use case is likely not supported because it was not foreseen back then.