siemens/kas #9: How to use template configs

zuka0828, 2019-06-05:
kas generates the `local.conf` and `bblayers.conf` in the build directory from the headers (`local_conf_header ` and `bblayers_conf_header`) and other kas variables (`machine`, `distro`, and the layers in `repos`).

Is it possible to use the existing template configs in each layer with kas?
i.e. `meta-xxx/conf/local.conf.sample` and `meta-xxx/conf/local.conf.sample`, that are copied by `oe-init-build-env` if we set `TEMPLATECONF=meta-xxx/conf`
I cannot find any information how to do this in `docs/userguide.rst`.

jan-kiszka, 2019-06-05:
Nope, this is not supported. The idea is so far to express artifacts contributing to the final local.conf via local_conf_header sections in the different kas files you may include per layer.

PS: we prefer to discuss on the mailing list.

zuka0828, 2019-06-30:
@jan-kiszka Thanks, I'd like to close the issue. It might be useful if kas supports loading existing `local.conf(.sample)` which would be provided in extension layers.

> PS: we prefer to discuss on the mailing list.

OK, I will from the next time.

closed 2019-06-30 by zuka0828