siemens/kas #11: kas(-docker) fails to detect proper path of isar source, when in git submodule

simonsudler, 2019-08-13:
When trying to build a isar based project, I noticed, that ``kas-docker`` has a problem when the git submodules.

When executing the same ``kas-docker`` build in a "normal" git repository, everything works fine. Executing the command in a git submodule fails to detect the proper git-root:

```
$ kas-docker --isar -v build kas/boards/some-project.yml
+ mkdir -p /home/user/isar/isar
+ docker run -v /home/user/isar/isar:/repo:ro -v /home/user/isar/isar:/work:rw --workdir=/work -e USER_ID=1000 -e GROUP_ID=1000 --rm -t -i -e SHELL='/bin/bash' -e TERM='xterm-256color' --cap-add=SYS_ADMIN --cap-add=MKNOD --privileged --device /dev/loop20 kasproject/kas-isar build /repo/kas/boards/some-project.yml
2019-08-13 11:10:04 - INFO     - kas 1.0 started
2019-08-13 11:10:04 - INFO     - /repo/kas$ git rev-parse --show-toplevel
2019-08-13 11:10:04 - INFO     - /repo/kas$ hg root
2019-08-13 11:10:04 - INFO     - /repo/kas/board$ git rev-parse --show-toplevel
2019-08-13 11:10:04 - INFO     - /repo/kas/board$ hg root
2019-08-13 11:10:04 - INFO     - /repo/kas$ git rev-parse --show-toplevel
2019-08-13 11:10:04 - INFO     - /repo/kas$ hg root
2019-08-13 11:10:04 - INFO     - Using /repo/kas as root for repository isar
2019-08-13 11:10:04 - INFO     - /work/isar$ git cat-file -t xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
```

``git rev-parse --show-toplevel`` command cannot work with reverence in the ``.git`` file, since the top-level repository is not mounted inside the docker container.

Why does ``kas`` require the ``git rev-parse --show-toplevel`` in the project root folder? It seams unnecessary for proper cloning of the meta-layers defined in the kas-file.

jan-kiszka, 2019-08-13:
It's needed to mount the repo, and only the repo. And only read-only. When you just specify an absolute path to a kas yml file, it's unclear where that repo start.

Can we follow-up the discussion on kas-devel@googlegroups.com?

jan-kiszka, 2019-08-20:
See https://groups.google.com/d/msgid/kas-devel/EB4794275C0C904BBABD6840AB8DC22D22A23A13%40DENBGAT9EL2MSX.ww902.siemens.net

closed 2019-08-20 by jan-kiszka