siemens/kas #14: patch entry looks wrong with several patches in the same repo

yuecelm, 2020-04-08:
```
2020-04-08 11:35:21 - INFO     - /work/isar-siemens$ git apply /repo/isar-patches/simatic-ipc-modules-uri.patch
2020-04-08 11:35:21 - INFO     - Patch applied. (patch path: /repo/isar-patches/simatic-ipc-modules-uri.patch, repo: isar-siemens, patch entry: simatic-ipc127e-static-ip)
:
2020-04-08 11:35:21 - INFO     - /work/isar-siemens$ git apply /repo/isar-patches/simatic-ipc127e-static-ip.patch
2020-04-08 11:35:21 - INFO     - Patch applied. (patch path: /repo/isar-patches/simatic-ipc127e-static-ip.patch, repo: isar-siemens, patch entry: simatic-ipc127e-static-ip)
```

my kas.yml:
```yaml
repos:
  isar-siemens:
    url: git@code.siemens.com:ebsy/debian/isar-siemens.git
    patches:
      simatic-ipc-modules-uri:
        repo: my-repo
        path: isar-patches/simatic-ipc-modules-uri.patch
      simatic-ipc127e-static-ip:
        repo: my-repo
        path: isar-patches/simatic-ipc127e-static-ip.patch
```

I also observed that kas try under some unknown condition to re-apply the simatic-ipc-modules-uri.patch again and fails the startup, but I cannot reproduce this case currently...

henning-schild, 2020-04-08:
Am Wed, 8 Apr 2020 04:57:44 -0700
schrieb yuecelm <notifications@github.com>:

> ```
> 2020-04-08 11:35:21 - INFO     - /work/isar-siemens$ git apply
> /repo/isar-patches/simatic-ipc-modules-uri.patch 2020-04-08 11:35:21
> - INFO     - Patch applied. (patch path: [...]

That looks fine to me. They have been applied.

> my kas.yml:
> ```yaml
> repos:
>   isar-siemens:
>     url: git@code.siemens.com:ebsy/debian/isar-siemens.git

Looks like you are missing the refspec. So kas does not know the
baseline those patches where applied on. You should try with a
commit-sha or a tag, giving it nothing or a branch name will not work.

> patches:
>       simatic-ipc-modules-uri:
>         repo: my-repo
>         path: isar-patches/simatic-ipc-modules-uri.patch
>       simatic-ipc127e-static-ip:
>         repo: my-repo
>         path: [...]

Very likely because of the missing refspec.

Henning

>

henning-schild, 2020-04-08:
This project does not use issues, please use the mailinglist instead kas-devel@googlegroups.com

Or use the issue tracker of the layer you are building on "isar-siemens"

closed 2020-04-08 by henning-schild

yuecelm, 2020-04-08:
I will do the next issue in isar-siemens, thanks for the hint

> ``` 2020-04-08 11:35:21 - INFO - /work/isar-siemens$ git apply /repo/isar-patches/simatic-ipc-modules-uri.patch 2020-04-08 11:35:21 - INFO - Patch applied. (patch path: [...]
right, both are applied, but both have `patch entry: simatic-ipc127e-static-ip`, is this really ok?

henning-schild, 2020-04-08:
I see, the entry ... sorry did not read it carefully since it was in the title and the body mentioned another issue.

The entry being the same twice is indeed weird and might be a bug in kas. And that should be reported to the kas mailinglist.

henning-schild, 2020-04-08:
I just ran the kas testsuite, it happens to apply two entries ... and their names are ok. Please include the kas version you are using when reporting this, and consider trying to reproduce with a more recent version if yours is old

yuecelm, 2020-04-08:
I am using kasproject/kas-isar:2.0, including kas release 2.0. let me do some more tests...

henning-schild, 2020-04-08:
Let us please stop talking here since this is not the official channel for support. Continue on the kas list or on your downstream layer. And please let me know why you need to patch the ipc-modules ... i am the maintainer of that repo inside Siemens

But not here, this should be the last message in this thread.