siemens/kas #15: use kas as a developer

nicola-lunghi, 2020-05-03:
Hi,
I am using meta-raspberrypi that is using kas.
1) I need to add a layer to the build and modify the kas file how can I do that without modifying the yaml file in meta-raspberrypy? 
2) as a developer I need to run a lot of bitbake commands is possible to launch the shell?

texierp, 2020-05-03:
Hi @nicola-lunghi ,

>     1. I need to add a layer to the build and modify the kas file how can I do that without modifying the yaml file in meta-raspberrypy?

The kas file in `meta-raspberrypi` is just a minimal example to work with kas, but
you can override it, see [1] and [2]. For further examples, you can check my dedicated
repo with some kas files ;) [3]

>     2. as a developer I need to run a lot of bitbake commands is possible to launch the shell?

Yes, you can do that with:

```
kas shell kas-files/kas-poky.yml:kas-files/board-raspberrypi3.yml -c "bitbake foo"
```

or

```
kas-docker shell kas-files/kas-poky.yml:kas-files/board-raspberrypi3.yml -c "bitbake foo"
```

If you use `kas-docker`.

FYI, this project does not use issues, please use the mailing-list instead kas-devel@googlegroups.com

/Pierre-Jean

[1] - https://kas.readthedocs.io/en/1.0/userguide.html#including-configuration-files-from-other-repos
[2] - https://kas.readthedocs.io/en/1.0/userguide.html#including-configuration-files-via-the-command-line
[3] - https://github.com/texierp/kas-files

nicola-lunghi, 2020-05-04:
Thanks!
I'll have a look at the kas-files project

jan-kiszka, 2020-07-01:
Looks like it was solved.

closed 2020-07-01 by jan-kiszka