siemens/kas #18: GitRepo fetch_cmd shouldn't default to "git fetch --all"

NGenetzky, 2020-07-16:
The fetch()  method for git will call git fetch --all ; this causes issues because I have 12+ forks added as remotes for one of my layers. It seems like this should be able to only fetch 'origin', which has been created by kas.

```python
class GitRepo(RepoImpl):
    """
        Provides the git functionality for a Repo.
    """
...
    def fetch_cmd(self):
        return ['git', 'fetch', '--all']
```

jan-kiszka, 2020-07-16:
Valid remark. I do not find any reason for the `--all` in the git histories.

Would you like to send this as patch with reasoning to kas-devel?

igaw, 2020-07-16:
My bad, it's my usual thing I do in the morning and I added it as command to kas.

NGenetzky, 2020-07-16:
> Would you like to send this as patch with reasoning to kas-devel?

I can, but it won't be immediate. I am still kind of new to the old school patch emailing collaboration approach.

- [ ] Create patch
- [ ] Submit patch to kas-devel
- [ ] Create PR
- [ ] Await feedback

jan-kiszka, 2020-07-16:
As the new-school web-based PR mechanisms are not inclusive to people that only work via mail clients, we have to restrict us to a single channel. I'd love to work with a platform that allows both ways, but neither github nor gitlab support that.

However, I can forward simple PRs to the list. Just make sure the commit is properly formatted (subject, message, signed-off - see https://github.com/siemens/kas/blob/master/CONTRIBUTING.md).

NGenetzky, 2020-07-25:
### 3 NGenetzky commented on xiUoNPQ0IaU

Submitting patch for review.

- [docs.google.com](https://docs.google.com/viewer?a=v&pid=forums&srcid=MTUxMDY3MjI4MDU2MzY4NzAwODIBMTAxMjM0ODMyMzE3NDA3OTI2NTkBYWtHdHhQcHpDQUFKATAuMQEBdjI&authuser=0)

jan-kiszka, 2020-07-27:
Applied to next.

closed 2020-07-27 by jan-kiszka