siemens/kas #20: Support 'matrix' builds

rossburton, 2020-07-17:
Say I have a base YAML file that but want to build it for a number of variations: six different machines, with musl or glibc, and two different distributions.

Currently that means 6*2*2 = 24 different yaml files would be needed.  It would be good if kas supported matrix builds, similar to Travis, so that these variations could be simply listed in a top-level yml that expands out to all of the required builds.

jan-kiszka, 2020-07-17:
[We prefer the mailing list for support]

Have a look at https://kas.readthedocs.io/en/latest/userguide.html#including-configuration-files-via-the-command-line.

NGenetzky, 2020-07-25:
> [We prefer the mailing list for support]

You might want to include the URL to the mailing list when you say that :smiley_cat: 

---

@rossburton , I think it's important to keep kas as simple as possible - and to leave the heavy lifting to bitbake. I would suggest looking into multilib. I would start with this post [Docs / Wind River Linux Platform Developer's Guide, 9 / Configuration and Build / Platform Project Image Configuration Overview / Creating Multiple Configuration Builds](https://docs.windriver.com/bundle/Wind_River_Linux_Platform_Developers_Guide_9_1/page/wkm1551818767221.html).

closed 2020-09-23 by jan-kiszka