siemens/kas #26: Silently fails if permission denied when setting up repositories

rossburton, 2020-10-13:
My kas in CI just failed with the following mysterious errors:

```
1202020-10-13 10:53:46 - INFO     - /builds/poky$ git remote set-url origin https://git.yoctoproject.org/git/poky
1222020-10-13 10:53:46 - INFO     - Changing remote URL to https://git.yoctoproject.org/git/poky failed with error: 
```

Note the absence of error messages.

The problem was that kas was running as a normal user but the `poky` checkout was owned by `root`.

jan-kiszka, 2020-10-13:
Do we get a proper message when that command is run in the shell? I do not spot yet where we should lose output.

jan-kiszka, 2020-10-13:
Sent a path to the list.

I suspect there are more cases of improper "fail=False" settings, but one after the other.

jan-kiszka, 2020-10-21:
merged to master

closed 2020-10-21 by jan-kiszka