siemens/kas #27: Kas changes Python search path

rossburton, 2020-10-16:
I've written some cunning/horrible code in Bitbake that uses modules on the host for improved logging.  Outside of Kas, this works fine.  Inside Kas, the imports fail.

jan-kiszka, 2020-10-16:
Please provide more details. And ideally use the mailing list for discussions.

NGenetzky, 2020-10-17:
@rossburton , please consider a minimal replication example next time.

Here is replication:

```
1d [ngenetzky:/work] 3s $ ( cat kas-minimal.sh  ; bash kas-minimal.sh )
#!/bin/bash
set -x

cat << EOF >> /tmp/kas.yml
header:
  version: 3
repos:
  poky:
    url: git://git.yoctoproject.org/poky
    refspec: "51f6145f8f99a02df1dad937684f014b0172e72a" # yocto-2.6.4 # thud
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:
EOF

PYTHONPATH='/tmp/bin/'
export PYTHONPATH
env > '/tmp/kas-before.env.sh'
kas shell '/tmp/kas.yml' -c 'env > /tmp/kas-after.env.sh'

grep 'PYTHONPATH' /tmp/kas-*.env.sh
+ cat
+ PYTHONPATH=/tmp/bin/
+ export PYTHONPATH
+ env
+ kas shell /tmp/kas.yml -c 'env > /tmp/kas-after.env.sh'
2020-10-17 01:37:12 - INFO     - kas 2.1.1 started
2020-10-17 01:37:12 - INFO     - /tmp$ git rev-parse --show-toplevel
2020-10-17 01:37:12 - INFO     - /tmp$ hg root
2020-10-17 01:37:12 - INFO     - /work/poky$ git remote set-url origin git://git.yoctoproject.org/poky
2020-10-17 01:37:12 - INFO     - /work/poky$ git cat-file -t 51f6145f8f99a02df1dad937684f014b0172e72a
2020-10-17 01:37:12 - INFO     - Repository poky already contains 51f6145f8f99a02df1dad937684f014b0172e72a as commit
2020-10-17 01:37:12 - INFO     - /work/poky$ git status -s
2020-10-17 01:37:12 - INFO     - /work/poky$ git rev-parse --verify HEAD
2020-10-17 01:37:12 - INFO     - 51f6145f8f99a02df1dad937684f014b0172e72a
2020-10-17 01:37:12 - INFO     - Repo poky has already been checked out with correct refspec. Nothing to do.
2020-10-17 01:37:12 - INFO     - /work/poky$ /tmp/tmpnw6w_nl_ /work/build
+ grep PYTHONPATH /tmp/kas-after.env.sh /tmp/kas-before.env.sh
/tmp/kas-before.env.sh:PYTHONPATH=/tmp/bin/
```

NGenetzky, 2020-10-17:
Conversation for this has moved to [Cannot import modules inside kas](https://groups.google.com/g/kas-devel/c/JXD7wQLu6Cg/m/u0q1uBDVBgAJ)

closed 2021-02-24 by jan-kiszka