siemens/kas #32: multiconfig located in top layer not found if using sub layers

HerrMuellerluedenscheid, 2020-11-18:
Hey
I refactored a project of to encapsulate some logic into a sub layer (lets say `meta-specific-config`). Thus it looks a bit like this:

```
meta-x:
  recipes ...
  meta-specific-config:
    recipes ...
```

Unfortunately kas, doesn't include the base layer recipes but only the `meta-specific-config` in the `bblayers` if I explicitly set the `layers` attribute in the kas configuration.
What my bblayers should contain for a specific configuration is:

```
  meta-x \
  meta-x/meta-specific-config \
```

How can I achieve that?

jan-kiszka, 2020-11-18:
```
layers:
  .:
  meta-specific-config:
```
should do the trick. If not, we need to fix something.

HerrMuellerluedenscheid, 2020-11-18:
That does the trick! Now that I see that I wonder why I didn't try before... Anyway, thanks!

closed 2020-11-18 by HerrMuellerluedenscheid

priv-kweihmann, 2021-05-05:
Coming back to this issue - indeed the proposed workaround does the trick **but** it results in 
```
meta-webserver       = "HEAD:945f062ff098dc9c8ba8d22c5eef88adec60730d"
.                    
meta-tpm             = "HEAD:3daf99fd138b0eebe864bbe1b9c71241d97c4512"
```

at the build console, which is directly translated into

```
meta-webserver    = HEAD:945f062ff098dc9c8ba8d22c5eef88adec60730d 
.                 = HEAD:3daf99fd138b0eebe864bbe1b9c71241d97c4512 
meta-tpm          = HEAD:3daf99fd138b0eebe864bbe1b9c71241d97c4512 
```

at /etc/build when using image-buildinfo bbclass - is there anyway to translate that item properly into the name.
BTW the shown example is when using meta-security and the embedded layer of it called meta-tpm

jan-kiszka, 2021-05-05:
How does the generated bblayers.conf look like, how should it look like?

priv-kweihmann, 2021-05-05:
Just send a patch, which fixes the issue for me - maybe we could continue the discussion on the devel-ML

priv-kweihmann, 2021-05-05:
but to give an answer to your question (and also as a public record)
```
<path>/meta-security/. \
<path>/meta-security/meta-tpm \
```

turns into

```
<path>/meta-security \
<path>/meta-security/meta-tpm \
```

which does look way better to me