siemens/kas #33: KeyError: 'getpwuid(): uid not found: 1000'

HerrMuellerluedenscheid, 2020-11-20:
Hey there,

I'm super happy with how kas brings a lot more structure to my build system. However, I'm stuck noch at about 60% build progress with the following exception which I do not get when not using my kas configuration.
I was wondering if you had a clue if I forgot to set some variable or where that might come from
```
NOTE: Using umask 0o002 (not 22) for sstate packaging
DEBUG: Executing python function sstate_hardcode_path
NOTE: Removing hardcoded paths from sstate package: 'find /home/pyrocko/yocto/qs-kas/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/kbd/2.2.0-r0/sstate-build-package/ \( -name "*.la" -o -name "*-config" -o -name "*_config" -o -name "postinst-*" \) -type f | xargs grep -l -e '/home/pyrocko/yocto/qs-kas/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/kbd/2.2.0-r0/recipe-sysroot' -e '/home/pyrocko/yocto/qs-kas/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/kbd/2.2.0-r0/recipe-sysroot-native' -e '/home/pyrocko/yocto/qs-kas/build/tmp/hosttools' | tee /home/pyrocko/yocto/qs-kas/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/kbd/2.2.0-r0/sstate-build-package/fixmepath | xargs --no-run-if-empty sed -i -e 's:/home/pyrocko/yocto/qs-kas/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/kbd/2.2.0-r0/recipe-sysroot-native:FIXMESTAGINGDIRHOST:g' -e 's:/home/pyrocko/yocto/qs-kas/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/kbd/2.2.0-r0/recipe-sysroot:FIXMESTAGINGDIRTARGET:g' -e 's:/home/pyrocko/yocto/qs-kas/build/tmp/hosttools:FIXME_HOSTTOOLS_DIR:g''
DEBUG: Python function sstate_hardcode_path finished
DEBUG: Executing python function sstate_report_unihash
WARNING: KeyError in .
ERROR: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:sstate_report_unihash(d)
     0003:
File: '/home/pyrocko/yocto/qs-kas/poky/meta/classes/sstate.bbclass', lineno: 840, function: sstate_report_unihash
     0836:    report_unihash = getattr(bb.parse.siggen, 'report_unihash', None)
     0837:
     0838:    if report_unihash:
     0839:        ss = sstate_state_fromvars(d)
 *** 0840:        report_unihash(os.getcwd(), ss['task'], d)
     0841:}
     0842:
     0843:#
     0844:# Shell function to decompress and prepare a package for installation
File: '/home/pyrocko/yocto/qs-kas/poky/bitbake/lib/bb/siggen.py', lineno: 555, function: report_unihash
     0551:
     0552:            if "." in self.method:
     0553:                (module, method) = self.method.rsplit('.', 1)
     0554:                locs['method'] = getattr(importlib.import_module(module), method)
 *** 0555:                outhash = bb.utils.better_eval('method(path, sigfile, task, d)', locs)
     0556:            else:
     0557:                outhash = bb.utils.better_eval(self.method + '(path, sigfile, task, d)', locs)
     0558:
     0559:            try:
File: '/home/pyrocko/yocto/qs-kas/poky/bitbake/lib/bb/utils.py', lineno: 420, function: better_eval
     0416:    if extraglobals:
     0417:        ctx = copy.copy(ctx)
     0418:        for g in extraglobals:
     0419:            ctx[g] = extraglobals[g]
 *** 0420:    return eval(source, ctx, locals)
     0421:
     0422:@contextmanager
     0423:def fileslocked(files):
     0424:    """Context manager for locking and unlocking file locks."""
File: '<string>', lineno: 1, function: <module>
  File "<string>", line 1, in <module>

File: '/home/pyrocko/yocto/qs-kas/poky/meta/lib/oe/sstatesig.py', lineno: 591, function: OEOuthashBasic
     0587:
     0588:                update_hash("\n")
     0589:
     0590:            # Process this directory and all its child files
 *** 0591:            process(root)
     0592:            for f in files:
     0593:                if f == 'fixmepath':
     0594:                    continue
     0595:                process(os.path.join(root, f))
File: '/home/pyrocko/yocto/qs-kas/poky/meta/lib/oe/sstatesig.py', lineno: 551, function: process
     0547:                    add_perm(stat.S_IXOTH, 'x')
     0548:
     0549:                if include_owners:
     0550:                    try:
 *** 0551:                        update_hash(" %10s" % pwd.getpwuid(s.st_uid).pw_name)
     0552:                        update_hash(" %10s" % grp.getgrgid(s.st_gid).gr_name)
     0553:                    except KeyError:
     0554:                        bb.warn("KeyError in %s" % path)
     0555:                        raise
Exception: KeyError: 'getpwuid(): uid not found: 1000'

DEBUG: Python function sstate_report_unihash finished
DEBUG: Python function sstate_task_postfunc finished

```
Thanks for taking a glimpse! :)

HerrMuellerluedenscheid, 2020-11-20:
I resolved the issue as follows:

    kas shell raspberrypi0-wifi-nied.yml -c 'bitbake -c cleanall gcc'
    kas shell raspberrypi0-wifi-nied.yml -c 'bitbake  gcc'

After that the build continued. I'm not sure about the root of that problem but when I built the project the first time, my hard drive was full. Probably something went wrong as a consequence.

closed 2020-11-20 by HerrMuellerluedenscheid

jan-kiszka, 2020-11-21:
Good that you were able to resolve this.

As an additional hint, this also works and is slightly shorter:
```
kas build raspberrypi0-wifi-nied.yml --target gcc -c cleanall
kas build raspberrypi0-wifi-nied.yml --target gcc
```