siemens/kas #35: ${AUTOREV} for `refspec`

HerrMuellerluedenscheid, 2020-12-04:
Hey,
I usually now work within our proprietary meta layer within the downloaded kas sources. I update things, run `kas build`, if everything works fine I update the kas configuration with that latest commit.
To enhance that workflow I was wondering if a `refspec=${AUTOREV}` given that a branch (or if not fallback to master) is provided wouldn't be an option.
Sure, explicit is better than implicit. On the other hand we already know (and I love) the `${AUTOREV}` feature from bitbake during development.

Best
Marius

jan-kiszka, 2020-12-04:
@pbrkr is using an auto-update model for CI purposes and can probably share some best practices in the kas context.
All our layer dependencies are currently pinned.

pbrkr, 2020-12-06:
@HerrMuellerluedenscheid: In a kas config file you can specify a commit hash, tag or even a branch in `refspec`.

A good example may be what I do in the meta-sancloud layer. We have a "rolling" configuration in `kas/dev/bbe-poky.yml` and `kas/dev/bbe-arago.yml` (the two distros we support) with `refspec: dunfell`, see https://github.com/SanCloudLtd/meta-sancloud/blob/dunfell/kas/dev/bbe-poky.yml for example. Then we have a "release" configuration in `kas/bbe-poky.yml` and `kas/bbe-arago.yml` which specifies fixed commit hashes to use for each layer, see https://github.com/SanCloudLtd/meta-sancloud/blob/dunfell/kas/bbe-poky.yml for example.

Does that help?

HerrMuellerluedenscheid, 2020-12-07:
@pbrkr awesome! Yes, that helps a lot!

closed 2020-12-07 by HerrMuellerluedenscheid

henning-schild, 2020-12-07:
Note that "branch" tracking is not supported. So using a branch in kas can get you stuck on a "random" commit ... depending on when you first ran kas to clone. So using branch-names as refspec is something you can do in CI ... where you hopefully start rather clean every single time. But for all other cases tags/branch-names are a bad idea, shas are the safe bet.

pbrkr, 2020-12-07:
> Note that "branch" tracking is not supported. So using a branch in kas can get you stuck on a "random" commit ... depending on when you first ran kas to clone. So using branch-names as refspec is [...]

This is why I added support for the `--update` and ``--force-checkout` arguments to kas :)

embetrix, 2022-12-03:
I would be nice if kas can provide an utility  to help setting the rolling refspec to fixed commits automatically when releasing.

fschrempf, 2022-12-14:
> I would be nice if kas can provide an utility to help setting the rolling refspec to fixed commits automatically when releasing.

Indeed, that would be helpful. I have the following maintenance helper tasks in mind:

* Bump the refspec of a repo set in the configuration to the latest HEAD of a given branch
* Write the currently used ref hashes to a new/existing YAML config file