siemens/kas #36: bblayers order

HerrMuellerluedenscheid, 2020-12-04:
Hey,
the order of layers in BBLAYERS in the generated build directory deviates from the order in my yml configuration.

After removing `bblayers.conf` (just to see that I'm not mistaken) I run `kas build mconfig.yml`. This is BBLAYERS:

```
BBLAYERS ?= " \
    /home/pyrocko/yocto/qs-kas/meta-mender/meta-mender-core \
    /home/pyrocko/yocto/qs-kas/meta-mender/meta-mender-raspberrypi \
    /home/pyrocko/yocto/qs-kas/meta-openembedded/meta-networking \
    /home/pyrocko/yocto/qs-kas/meta-openembedded/meta-oe \
    /home/pyrocko/yocto/qs-kas/meta-openembedded/meta-python \
    /home/pyrocko/yocto/qs-kas/meta-openembedded/meta-webserver \
    /home/pyrocko/yocto/qs-kas/meta-quakesaver/. \
    /home/pyrocko/yocto/qs-kas/meta-quakesaver/meta-qs-raspberrypi \
    /home/pyrocko/yocto/qs-kas/meta-scipy \
    /home/pyrocko/yocto/qs-kas/poky/meta \
    /home/pyrocko/yocto/qs-kas/poky/meta-poky \
    /home/pyrocko/yocto/qs-kas/poky/meta-yocto-bsp"
 
```
And this is the order of my yml file:

```
  meta-quakesaver:
    layers:
      .:
      meta-qs-raspberrypi:

  meta-scipy:

  meta-mender:
    layers:
      meta-mender-core:
      meta-mender-raspberrypi:
      meta-mender-core:

  meta-openembedded:
    layers:
      meta-oe:
      meta-python:
      meta-networking:
      meta-webserver:

  poky:
    layers:
       meta:
       meta-poky:
       meta-yocto-bsp:

```
As you can see it's not reversed but just different. The actual config includes another config.yml but that one doesn't define any layers or repos. Any idea why that happens?

HerrMuellerluedenscheid, 2020-12-04:
BTW: I removed `url` and `refspec` to make the example a little more lightweight.

jan-kiszka, 2020-12-04:
kas processes repos in parallel, plus repos and layers can come different include files. This made generated config files undeterministic, see c9326cc1edc9e84e950cdec7f0cecd4a520570ac. I'm open for a more user-controllable ordering algorithm, provided it's deterministic. But it's not a simple thing to solve I'm afraid.

Weren't layer prior invented for sorting out such ordering issues?

HerrMuellerluedenscheid, 2020-12-05:
I'm not entirely sure and probably lack deeper understanding of how things are put together under the hood. I always thought that layer order actually matters. Given that two layers have identical priorities wouldn't assignments with e.g. `?=` result in different settings dependent on the ordering of layers?

I asked in the yocto irc if order matters and basically got two replies:

> Hey there: just to confirm: the order of the layers defined in BBLAYERS matters, right?
> mckoan: Yes
> mckoan: 1st priority, 2nd order
> qschulz: it's more complex than that, so if you have a precise question we can answer
> qschulz: bbclass and conf files are not handled the same way as bb or bbappends

If order matters, c9326cc1edc9e84e950cdec7f0cecd4a520570ac would be rather counterproductive.

jan-kiszka, 2020-12-06:
That commit as needed as we were otherwise producing even random orders. But things seem to have changed since then, including `dict` becoming ordered by default. I think we can try reverting this. Can you check how far this gets us?

```
diff --git a/kas/config.py b/kas/config.py
index 0119f68..9b827fb 100644
--- a/kas/config.py
+++ b/kas/config.py
@@ -24,6 +24,7 @@
 """
 
 import os
+from collections import OrderedDict
 from .repos import Repo
 from .includehandler import IncludeHandler, IncludeException
 
@@ -87,7 +88,7 @@ class Config:
         """
         repo_config_dict = self._config.get('repos', {})
         repo_defaults = self._config.get('defaults', {}).get('repos', {})
-        repo_dict = {}
+        repo_dict = OrderedDict()
         repo_fallback_path = os.path.dirname(self.filenames[0])
         for repo in repo_config_dict:
             repo_config_dict[repo] = repo_config_dict[repo] or {}
@@ -128,7 +129,7 @@ class Config:
             Returns the local.conf header
         """
         header = ''
-        for key, value in sorted(self._config.get(header_name, {}).items()):
+        for key, value in self._config.get(header_name, {}).items():
             header += '# {}\n{}\n'.format(key, value)
         return header
 
diff --git a/kas/libcmds.py b/kas/libcmds.py
index d65048d..750c973 100644
--- a/kas/libcmds.py
+++ b/kas/libcmds.py
@@ -245,8 +245,8 @@ class WriteBBConfig(Command):
                 fds.write(ctx.config.get_bblayers_conf_header())
                 fds.write('BBLAYERS ?= " \\\n    ')
                 fds.write(' \\\n    '.join(
-                    sorted(layer for repo in ctx.config.get_repos()
-                           for layer in repo.layers)))
+                    layer for repo in ctx.config.get_repos()
+                    for layer in repo.layers))
                 fds.write('"\n')
 
         def _write_local_conf(ctx):
```

We may still have an undesired order of things across includes, simply because ordering was so far not checked. If that's the case, I suppose we can sort that our before the next release if folks with such workload provide feedback (our layering does not rely on ordering, thus cannot help).

jan-kiszka, 2020-12-14:
Ping @HerrMuellerluedenscheid

HerrMuellerluedenscheid, 2020-12-14:
Hey, sorry for the late reply. I did a quick test and can confirm that that patch preserves the layer order!

HerrMuellerluedenscheid, 2020-12-14:
Let me know if I can help with anything else on that end.
Feel free to close the issue.

jan-kiszka, 2021-02-24:
Sorting by list order was discussed in https://groups.google.com/g/kas-devel/c/kbX_NAtQEMk, and we concluded to stick with ordering by name.

closed 2021-02-24 by jan-kiszka

paweljonskim, 2021-03-19:
Hello,
unfortunately ordering layers by name makes kas unusable in our environment. We cannot really change that because of external yocto environment delivered from another company, but we still would like to use with kas. Is it closed matter or maybe we could find a way to maintain the yml layer order somehow?

jan-kiszka, 2021-03-19:
Ordering is by layer **key**, as defined in the kas files. Is that external delivery providing you kas files as well? But even then: There is always a way to move a string before or after an existing set of stings.

paweljonskim, 2021-03-19:
Can you please explain what do you mean by _layer key_?
No, the provided repo does not contain kas config file.
I will elaborate my case. Part of kas config yml file:
`repos:
  repo_name:
    url: "git@github.com:org/repo_name.git"
    refspec: master
    layers:
      meta-c:
      meta-b:
      meta-a:
`
ends up creating bblayers.conf:
`
BBLAYERS ?= " \
    /home/pjonski/workspace/kasproject/repo_name/meta-a \
    /home/pjonski/workspace/kasproject/repo_name/meta-b \
    /home/pjonski/workspace/kasproject/repo_name/meta-c \
`
So repo names get sorted, which is not what I would expect. What do you mean by _layer key_?

jan-kiszka, 2021-03-19:
Please read the email thread referenced in this issue, specifically https://groups.google.com/g/kas-devel/c/kbX_NAtQEMk/m/jYQzuz5VAgAJ.

90degs2infty, 2024-03-13:
Sorry for resurrecting the dead. I've just stumbled upon a similar problem and wanted to add some thoughts on this.

From how I read this issue, `kas` relies on the layer ordering being specified by means of priorities (i.e. out of control of `kas`, as the layers themselves should set [`BBFILE_PRIORITY_...`](https://docs.yoctoproject.org/ref-manual/variables.html#term-BBFILE_PRIORITY) as needed). Consequently, the way `kas` orders the layers when writing `BBLAYERS` to `conf/bblayers.conf` _should_ not matter. However, from how I understand Yocto's docs, the ordering in fact _does_ matter:

> Priority values control which layer takes precedence if there are recipe files with the same name in multiple layers.
> For these cases, the recipe file from the layer with a higher priority number takes precedence.
> Priority values also affect the order in which multiple `.bbappend` files for the same recipe are applied.
>
> -- [docs.yoctoproject.org](https://docs.yoctoproject.org/dev-manual/layers.html#prioritizing-your-layer)

Note in the above, that the docs explicitly refer to _recipes and append files_.
Quoting the docs some more:

> Also, the layer priority does not currently affect the precedence order of `.conf` or `.bbclass` files.
> Future versions of BitBake might address this.
>
> -- [docs.yoctoproject.org](https://docs.yoctoproject.org/dev-manual/layers.html#prioritizing-your-layer)

and

> BitBake parses each `conf/layer.conf` file from the top down as specified in the [BBLAYERS](https://docs.yoctoproject.org/ref-manual/variables.html#term-BBLAYERS) variable within the `conf/bblayers.conf` file.
> During the processing of each `conf/layer.conf` file, BitBake adds the recipes, classes and configurations contained within the particular layer to the source directory.
>
> -- [docs.yoctoproject.org](https://docs.yoctoproject.org/dev-manual/layers.html#enabling-your-layer)

So to my understanding, `bitbake` respects priorities only in some context, while in others it falls back to the order in `BBLAYERS`.

In my specific use case, it was enough to employ the workaround referenced [above](https://github.com/siemens/kas/issues/36#issuecomment-802766510). But I feel like this only works if there is a one-to-one mapping between git-repositories and layers. As soon as one maintains several layers within the same git repository,[^1] there is no way for the user to fully control the layers' ordering in `BBLAYERS`, right? So as long as `bitbake` in parts relies on the ordering in `BBLAYERS`, it might be beneficial to provide a fully user-controlled way to order layers from within `kas`?

[^1]: as is the case e.g. for [`meta-openembedded`](https://github.com/openembedded/meta-openembedded)

katomaso, 2025-01-24:
Hi, I am new to KAS and unfortunately, this was the first issue I stumbled upon. The current ordering is very confusing. Why not to honor the Yocto ordering expectations and have a deterministic ordering by simply ordering by `BBFILE_PRIORITY`?

Ordering by `BBFILE_PRIORITY` would make this hidden value de-facto visible in `bblayers.conf` and would be perfectly in line with Yocto expectation.

fmoessbauer, 2025-01-24:
Hi @katomaso , Yocto and their layer precedence seems to be a never ending story. For backwards compatibility we can't simply change the ordering, but what we could think of is a `layer_order` top-level key that can be set to `priority`.

Once we have that option we can come up with a simple parser to extract the value of `BBFILE_PRIORITY_<foo>` and order by priority and name (given that two layers have the same priority, which is quite common).

Would that solution work for you?

jan-kiszka, 2025-01-24:
If we want to give control over the ordering in BBLAYERS, that should be made explicit, e.g. by adding an optional key to the layer objects:
```
layers:
  meta-begin:
    prio: 99
  meta-middle:
  meta-end
    prio: -99
```

katomaso, 2025-01-28:
Hi @fmoessbauer , your solution seem ideal. Less configuration is better but one `layer_order` for the sake of backward compatibility is necessary.  I hope this would finally settle this issue.

c92s, 2025-09-17:
Sorry to bump this old thread, but are there plans to add such a `layer_order` or `priority` in future? Adapting the layer path or names just for orderings sake feels kind of awkward here. 
I just stumbled across this issue when using `meta-arm-bsp` and `oe-core` which have both priority set to 5...

jan-kiszka, 2025-09-17:
Needs someone who feels enough pain to work on this. We would support and guide this, of course.

fmoessbauer, 2026-01-20:
With 47570d74104b62514ee0e31d72ea69a1ce7a27eb being merged now, this issue should finally be resolved. The change will then be part of the next (5.2) release.