siemens/kas #40: Add support for custom build directory

kubasejdak, 2021-01-07:
Hi,

It seems that kas has hardcoded `build` as default name for the build directory inside working directory.
It is quite problematic if we want to build multiple images from the same set of layer repositories.

For example, I have custom layer with local changes and I want to test it with multiple target images (e.g. for different machines). I am unable to do that with kas.

Some solution would be to provide an extra cmd line argument `--build` where I could specify what should be the BitBake workdir. This way paths where repositories are cloned/stored and build dir are separated.

Or maybe there is some mechanism to do that already?
Thanks!

jan-kiszka, 2021-01-07:
As you correctly found out, the build dir is always `${KAS_WORK_DIR}/build` at this point. You can reuse common checkouts via `KAS_REPO_REF_DIR`: First image is build normally into `workdir-1`, second image is built into `workdir-2`, setting `KAS_REPO_REF_DIR` to `workdir-1`. That's for the theory, I'm personally not using that, so please let us know if there are regressions.

If that mechanism is insufficient, you may suggest overwriting the build dir via another env var, but then this needs some reasoning.

jan-kiszka, 2021-01-07:
Err, no, that repo-ref requires the repos to be named in a specific way. Better ask on the mailing list for the intended workflow, CC'ing Claudius Heine who once wrote this feature.

jan-kiszka, 2021-02-24:
I've seen no follow-up on the ML. Resolved otherwise?

phatina, 2021-02-24:
Hi, can we achieve the original intent by setting `KAS_WORK_DIR` to something else?

jan-kiszka, 2021-02-25:
If you read my replies, you can see that `KAS_WORK_DIR` alone will not help. Reusing layer checkouts may work via `KAS_REPO_REF_DIR`, but how exactly should be clarified with the original author.

kubasejdak, 2021-02-25:
I can live without it, so it only my opinion is stopping from closing, then please do it. Or I can if you want me to.

jan-kiszka, 2021-02-25:
The offer to discuss the use case on the mailing list still stands. Closing here.

closed 2021-02-25 by jan-kiszka