siemens/kas #42: kas-container usage

coschneider, 2021-01-29:
Hi,

thanks for your great work.

When I try to run the following command I get an error :
```bash
$ KAS_WORK_DIR=$PWD/tmp-build/ DL_DIR=$PWD/downloads ./kas-container shell test.yml -c "bitbake package-index"
usage: kas [-h] [--version] [-d] {build,checkout,for-all-repos,shell} ...
kas: error: unrecognized arguments: /repo/test.yml
```

How can I call a shell bitbake command with the docker container solution? With plain kas command it works as expected.

jan-kiszka, 2021-01-29:
Please try out https://groups.google.com/g/kas-devel/c/E3mv6OCN_X8 and give feedback if that works.

coschneider, 2021-01-29:
Now it works like a charm. Thanks a lot for your quick fix/patch.

commit d647727f (2021-01-29): kas-container: Add support for multi-word --command arguments

Yet another special dance needed to preserve argument boundaries while
moving them around: This trick enables

kas-container shell kas.yml --command "echo it works."

by keeping the command argument separate and injecting it via
appropriate quoting into the argument array.

Closes: #42
commit 4042ef48 (2021-01-29): kas-container: Add support for multi-word --command arguments

Yet another special dance needed to preserve argument boundaries while
moving them around: This trick enables

kas-container shell kas.yml --command "echo it works."

by keeping the command argument separate and injecting it via
appropriate quoting into the argument array.

Closes: #42
closed 2021-02-19 by jan-kiszka