siemens/kas #44: External Compiler/Tool Support

mustafagonul, 2021-02-16:
Hi,

I am trying to use kas to build yocto and I am experimenting building TI Processor SDK with kas. You clan find related information via the link below:

https://software-dl.ti.com/processor-sdk-linux/esd/docs/latest/linux/Overview_Building_the_SDK.html

For building the SDK, two external compilers are used:

```
$ wget https://developer.arm.com/-/media/Files/downloads/gnu-a/8.3-2019.03/binrel/gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf.tar.xz
$ tar -Jxvf gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf.tar.xz -C $HOME
$ wget https://developer.arm.com/-/media/Files/downloads/gnu-a/8.3-2019.03/binrel/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu.tar.xz
$ tar -Jxvf gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu.tar.xz -C $HOME
```

and some exports are done: 

```
$ export TOOLCHAIN_PATH_ARMV7=$HOME/gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf
$ export TOOLCHAIN_PATH_ARMV8=$HOME/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu
```

I did not know how to add into the yaml file, it would be nice if we can support that.

Kind regards,
Mustafa

jan-kiszka, 2021-02-16:
Two options, I would say:
- derive your own kas container that has those binaries pre-installed, like we installed the debian toolchain
- model fetching and installation as bitbake target and let the SDK recipes depend on it (if that it possible, never done that myself)

See also #31 for a similar discussion.

mustafagonul, 2021-02-16:
Hi Jan,

While I was discussing the issue with my colleague, I have decided to update the docker and add the necessary stuff into that. Previously I was building the SDK with script inside a docker container (I have also a custom dockerfile.), May be, I can combine these two.

Thanks,
Mustafa

closed 2021-02-16 by mustafagonul

henning-schild, 2021-02-16:
In fact a simple self-contained thing like that probably best is mounted as volume or layered on top of the central container, no need to "fork" that container