siemens/kas #45: Relative layer path gets treated as absolute path

manuel-wagesreither, 2021-02-25:
Hello there!

I'd like my bblayers.conf to include a layer which lives in a subdirectory of my repository. However, the layer gets referenced in bblayers.conf in an absolute way.

In `kas.yml` I define the layer to live in subdirectory `meta-manwag` of the repository:
```
repos:
  meta-manwag:
    path: meta-manwag
```

But the generated bblayers.conf turns it into an absolute path.
```
BBLAYERS ?= " \
    [...]
    /home/manuel/bora-proj/poky/meta-poky \
    /home/manuel/bora-proj/poky/meta-yocto-bsp \
    meta-manwag"
```

The [documentation](https://kas.readthedocs.io/en/latest/userguide.html#configuration-reference) says the following regarding the path keyword:
> The path where the repository is stored. If the url and path is missing, the repository where the current configuration file is located is defined. If the url is missing and the path defined, this entry references the directory the path points to. If the url as well as the path is defined, the path is used to overwrite the checkout directory, that defaults to kas_work_dir + repo.name. In case of a relative path name kas_work_dir is prepended.

As per [documentation](https://kas.readthedocs.io/en/latest/command-line.html#environment-variables) `KAS_WORK_DIR` shall default to the current work directory. I tried setting it explicitely (`KAS_WORK_DIR=$( pwd) kas shell my.yml`), but that didn't help.

This problem also shows with ghcr.io/siemens/kas/kas:2.3.3. Strangely, I have some other kas container with digest `sha256:7772f9bfe9db3e61effdb8ef76dbd0b2d778668dc6d1581cb76cf0363046d4c2` in my repository, which works just fine. I yet have to find a way to diff it with ghcr.io/siemens/kas/kas:2.3.3.

If you want, you can recreate this problem in the project it occured:

1. `git clone https://gitlab.com/manuel_wagesreither/bora-proj.git`
2. `kas shell kas/bora-qemux86-64.yml`

Thank you for creating kas! Managing layers with google repo tool was horrible.
Best regards,
Manuel

henning-schild, 2021-02-25:
```
repos:
  this:
  foo:
    url: foo.org/bar
    refspec: 0815
    layers:
        bla/bob
        foo
```

henning-schild, 2021-02-25:
did not read your whole post, but that is how you select some layers that live in sub-folders of repos, can be multiple ... by default a repo is taken as one layer which does not seem what you have

henning-schild, 2021-02-25:
https://kas.readthedocs.io/en/latest/userguide.html?highlight=layers#configuration-reference

manuel-wagesreither, 2021-02-25:
@henning-schild 

When I change my kas.yml to the following:
```
repos:
  meta-manwag:
  meta-seconddir:
  meta-thirddir:
  meta-openembedded:
    url: "git://git.openembedded.org/meta-openembedded"
    refspec: 144077083457ccb75700777d5e6b1da9aaff01aa
    layers:
      meta-oe:
      meta-multimedia:
      meta-networking:
      meta-python:
    path: meta-openembedded
```
this is what I get:
```
BBLAYERS ?= " \
    /home/manuel/bora-proj \
    /home/manuel/bora-proj/ \
    /home/manuel/bora-proj/ \
    /home/manuel/bora-proj/meta-openembedded/meta-multimedia \
    /home/manuel/bora-proj/meta-openembedded/meta-networking \
    /home/manuel/bora-proj/meta-openembedded/meta-oe \
    /home/manuel/bora-proj/meta-openembedded/meta-python \
   [...]
```

> did not read your whole post, but that is how you select some layers that live in sub-folders of repos, can be multiple ... by default a repo is taken as one layer which does not seem what you have

That's not what I want. I want to select some directories of the local directory as layers for bitbake.

Let this be my directory layout:
```
kas.yml
meta-manwag
meta-firstdir
meta-seconddir
README.md
[...}
```
I'd expect kas to clone meta-openembedded and put it there:
```
kas.yml
meta-manwag
meta-firstdir
meta-seconddir
meta-openembedded <---- got cloned by kas from the internet
README.md
[...}
```
and that's the bblayers I'd expect from the documentation:
```
BBLAYERS ?= " \
    /home/manuel/bora-proj/meta-manwag \
    /home/manuel/bora-proj/meta-seconddir \
    /home/manuel/bora-proj/meta-thirddir \
    /home/manuel/bora-proj/meta-openembedded/meta-multimedia \
    /home/manuel/bora-proj/meta-openembedded/meta-networking \
    /home/manuel/bora-proj/meta-openembedded/meta-oe \
    /home/manuel/bora-proj/meta-openembedded/meta-python \
```

henning-schild, 2021-02-25:
You seem to have multiple repos without refspec etc.

If those layers are in your uppermost-repo already ... that guy will need a "layers", also if those other layers are coming in with i.e. submodules, "repo" or what you might have.

```
repos:
 your-funny-uppermost-name:
  layers:
    meta-manwag
    meta-firstdir
    meta-seconddir
 oe:
  ...
```

manuel-wagesreither, 2021-04-11:
Solved my issue, thanks!

@henning-schild
So by leaving out `path` it gets set to the current working directory, and the subdirs get handled as layers within that repo. I see! That didn't occur to me.

closed 2021-04-11 by manuel-wagesreither