siemens/kas #48: Docker image version is out-of-date

vivien, 2021-03-18:
Kas latest release is 2.3.3 but the latest Docker image is 2.2. Can a 2.3.3 docker image be published?

jan-kiszka, 2021-03-19:
In fact, latest is 2.4.

You likely still look at old dockerhub, but we migrated here, to ghcr.io. See https://kas.readthedocs.io/en/latest/userguide.html#usage.

closed 2021-03-19 by jan-kiszka

vivien, 2021-03-19:
Nope, this [Usage](https://kas.readthedocs.io/en/latest/userguide.html#usage) documentation provides a [link](https://github.com/orgs/siemens/packages/container/kas%2Fkas/31765) only mentioning kas:2.2.

jan-kiszka, 2021-03-19:
Ah, indeed, thanks! We need to patch in some better link, maybe https://github.com/orgs/siemens/packages/container/kas%2Fkas/versions. Let me check...