siemens/kas #51: for-all-repos using `kas-container`

rotem443, 2021-05-16:
I'm trying to run this following command in order to get all the repositories used:
`./kas-container.sh for-all-repos 'echo "REMOTE_URL: ${git ls-remote --get-url}"' {kas_config}`

unfortunately this feature is not supported yet, I have tried to implement the following feature but it seems a bit more complicated then I have anticipated.
The first reason is that the parsing order is different on `for-all-repos` command

**build** - kas build  [-c TASK] config
**shell** - kas shell  [-c COMMAND] config
**for-all-repos** - kas for-all-repos config command

So some new flags/some hacks required to make them work together. or a bit deeper change rather than just adding a param.
The other issue I have encountered is that for some reason I can't run for-all-repos without the `-d` flag

Example:
> docker run ghcr.io/siemens/kas/kas:2.4 for-all-repos
>> error: exec: "for-all-repos": executable file not found in $PATH
 
> docker run ghcr.io/siemens/kas/kas:2.4 -d for-all-repos
>> usage: kas for-all-repos [-h] [--skip SKIP] [--force-checkout] [--update]
                         config command
kas for-all-repos: error: the following arguments are required: config, command

I would be happy to implement the following feature with some guildness. Thanks!

jan-kiszka, 2021-05-17:
True, and we also lack `checkout` support in kas-container.

I would expect that both commands "only" need resource and parameter forwarding, similar to `build`. Try adding them so that they are recognized, including their own parameters, and see how far you get. Please share you results, even if intermediate, with the mailing list to have everyone informed.

rotem443, 2021-05-18:
Thanks @jan-kiszka I would try to work on this later on. For now, I'm doing this in a hacky way of running the for-all-repos with 
`kas-container.sh shell -c "/kas/run-kas for-all-repos"....`

jan-kiszka, 2021-05-20:
Just pushed an experimental patch to 'next'. You will both need the new script as well as kas[-isar]:next for this to work. Please test and report.

rotem443, 2021-05-20:
Great! Thanks! It seems to work.
The only issue that I'm still facing is that it's working only with the "-d" flag, I'm not sure if this is a problem on my side

`./kas-container -d --isar for-all-repos kas/config.yml "git remote -v"` --> working
`./kas-container -isar for-all-repos kas/config.yml "git remote -v"`  --> error: exec: "for-all-repos": executable file not found in $PATH

jan-kiszka, 2021-05-20:
As I wrote, you also need to pull the latest (:next) container image.

rotem443, 2021-05-20:
Sorry my bad, it's working without the debug flag, thanks!

commit 1c9e429e (2021-05-20): kas-container: Add support for checkout and for-all-repos

Allow invoking those two new sub-commands also via the container
interface.

Closes #51
commit 46eabc7b (2021-05-20): kas-container: Add support for checkout and for-all-repos

Allow invoking those two new sub-commands also via the container
interface.

Closes #51
closed 2021-05-24 by rotem443