siemens/kas #53: export the 'manifest'

vermaete, 2021-05-26:
I'm looking for the same functionality of 'repo manifest -r -o my-manifest.xml' of google repo in Kas.
Is there something to export all git hash stuff from the meta layers into a new exported YML file?

jan-kiszka, 2021-05-27:
I suppose the scenario is having either an include combination on the command line (`kas.yml:opt.yml...`) or an include hierarchy inside the yaml files, right? What would be the use case in the end?

In any case, we do not yet have a re-export of a flattened yaml file yet. Relates a bit to what we are discussing on the mailing list: https://groups.google.com/d/msgid/kas-devel/76b80bb2-3792-528d-5ed9-6c15b969a194%40siemens.com.

vermaete, 2021-05-27:
Hi Jan,

I thing that a flattened yml file is what I'm looking for.

Use case:
 1 ) CI starts with the normal yml file that in in SCM.
 2) Once it's done, it will export a new yml file with the hashes of all git repo's hard coded.  E.g. 'release.yml'.
 3) And CI put's a git tag on that file.
 4) ISO auditor comes and believes everything is under control to rebuild this exact image in a galaxy far far away.

This is now working for use with google repo with 'repo manifest -r -o release.xml'.
But nobody would be missing repo if this use case would work with KAS.

 br

jan-kiszka, 2021-05-27:
OK, so you have another requirement on top: not only unflatten the yaml files, but also translating floating branch refs into SHAs, right? Technically surely feasible. Would likely be best modeled as additional parameter to `kas build`. Now it just takes someone to sit down and code this...

jan-kiszka, 2022-08-26:
Still relevant? Anyone interested in working on this?

chrysh, 2022-10-28:
I wrote a python script for our use case, maybe it's relevant to somebody: https://github.com/chrysh/kas-scripts

fmoessbauer, 2022-11-06:
@vermaete, @chrysh : Please have a look at this patch series [1]. I guess it handles all your requirements:

- flatten all includes into one file (json or yaml), no matter how they are specified (include, vs. cmdline)
- relative refspecs can be resolved (e.g. branch name -> SHA)
- output itself is kas compatible (only yaml version) and without any loss of semantic information

[1] https://groups.google.com/g/kas-devel/c/_SRmcWulI6I

jan-kiszka, 2023-01-12:
IMHO, we have this feature now in latest kas. If something is missing, reopen or create a new issue.

closed 2023-01-12 by jan-kiszka

jan-kiszka, 2023-01-13:
Please also note that Felix suggest a lock extension now, see https://groups.google.com/g/kas-devel/c/pY8G4JuaQS8. Any feedback on that should go to that posting, please.

timblaktu, 2024-10-10:
I couldn't get @chrysh's script to run on python3, and arrived at this command to convert a repo manifest to yaml, which can be easily and more importantly automatically, massaged into a proper kas yaml. (Note, use the golang yq, not the older python one...)

```
yq -p=xml -o=yaml --xml-attribute-prefix '' --xml-skip-proc-inst '.' variscite-bsp-platform/imx-6.1.36-2.1.0.xml > variscite-bsp-platform/imx-6.1.36-2.1.0.yml
```

As you can see, I'm building for a variscite SoM right now. I put the above into a Makefile as a pre-proc step so that I can automatically accept upstream changes to variscite-bsp-platform. This step seems critical, since many/most of the hardware providers are supporting kas build tools yet. I tried to sell them on it, mostly bc they had no idea what it was, not that kas shouldn't already sell itself to such an org churning out BSP versions furiously.

Does anyone (@jan-kiszka) know whether there is another automation out there to convert repo manifests to kas yaml? Seems like something that would behoove the kas team to focus on for more adoption, but I couldn't find it.

jan-kiszka, 2024-10-11:
@timblaktu A converter form repo xmls to kas configs would be a separate topic. Maybe you can create a new issue for that.

In addition, asking around on the mailing list may bring forward the experience of other users with this (if there is any around auto-conversion).

fmoessbauer, 2024-10-11:
@timblaktu Converting the repo parts from the Manifest to a kas repo section is straight forward. I just created a [simple script](https://gist.github.com/fmoessbauer/3447e5eee62f874f2ceb59da384d4c4c) to do this. However, we lack the information about the layers. If you have an idea how do get this (e.g. by parsing a local `bblayers.conf`), that would be appreciated.

The script is not much tested yet, but at least works on the `variscite-bsp-platform` manifest so that kas can checkout the repos. Anyways, please move this discussion to the ML to have a broader audience.

timblaktu, 2024-10-11:
@fmoessbauer that script looks like the one posted earlier by @chrysh. I started there, but was not willing to spend any time on it after encountering many early import errors. I have implemented this in shell script, basically just some calls to jq and yq, and it is doing the right thing, minus proper yamlification of bblayers, which @fmoessbauer mentioned. I will start a new issue on this specific feature request, and a related conversation on the google group..

timblaktu, 2024-10-11:
https://groups.google.com/g/kas-devel/c/Dk2AKNx0PQA