siemens/kas #60: Docker image outdated

HerrMuellerluedenscheid, 2021-09-02:
I saw that your docker images have not been updated in a year which can be a security risk for people unsing that image. Would be better to have the docker build and deploy somehow hooked into your github actions.

HerrMuellerluedenscheid, 2021-09-02:
Well, okay, its in your actions but it should regularly rebuild. Would make things more secure for many people.

jan-kiszka, 2021-09-02:
I suspect you are starring at the legacy repo on "good-old" dockerhub. We are on ghcr.io now. See README.

HerrMuellerluedenscheid, 2021-09-02:
I see. Still, dockerhub is for many (including me) to go-to lookup place. Can you drop that image? Also that would be much safer than letting that rot there ;)

jan-kiszka, 2021-09-02:
I don't want to drop it to avoid breaking older kas-docker scripts.

https://hub.docker.com/r/kasproject/kas/
```
kas builder image (NOTE: project migrated to ghcr.io/siemens/kas/kas)
```
If you can suggest a way to make the printing of that sign better visible, I'm all ears, though.

henning-schild, 2021-09-02:
Like most containers it unfortunately is not maintained security wise. That is totally unrelated to the legacy drop location (dockerhub). The more frequent we update (for other reasons) the less CVEs will be in there, but CVEs at the moment are no reason to push updated containers.

With docker introducing a pricing model, we can suspect that dockerhub will become less of a default lookup place for many.

This repo contains the Dockerfiles, anyone is free to self-build. In which case an "apt-get upgrade" would probably be handy to also update lower layers ... because debian itself will not release new container images only because some CVEs have been fixed in a package.

henning-schild, 2021-09-02:
@jan-kiszka we could push one final container that includes a big fat deprecation warning in its entrypoint

```
echo "WARNING: This container image is not maintained, switch to gh"
echo "WARNING: Giving you time to think about that"
sleep 300
```

HerrMuellerluedenscheid, 2021-09-02:
@henning-schild I think that is a good idea. On top of that I would add a big fat deprecation warning to the dockerhub landing page with a reference to the new location.

HerrMuellerluedenscheid, 2021-09-02:
> I don't want to drop it to avoid breaking older kas-docker scripts.
> 
> https://hub.docker.com/r/kasproject/kas/
> 
> ```
> kas builder image (NOTE: project migrated to ghcr.io/siemens/kas/kas)
> ```
> 
> If you can suggest a way to make the printing of that sign better visible, I'm all ears, though.

I just saw that little notification there. Well, I guess I didn't read properly. I'm the living proof that this is easy to overlook, though.

henning-schild, 2021-09-02:
Here is an example with a little more prominent deprecation notice ... https://hub.docker.com/_/opensuse

Essentially we could have something about deprecation in the place where we just have the link to github.

jan-kiszka, 2021-09-02:
Did some editing, hope it's better. I still don't want to touch images, though.

HerrMuellerluedenscheid, 2021-09-02:
Thanks @jan-kiszka ! Yes I think that is much more prominent

closed 2021-09-02 by HerrMuellerluedenscheid