siemens/kas #75: Add support for environment variables support for YAML file

coldspark29, 2022-01-04:
I need to expand an environment variable in one of the repository urls, but kas doesn't seem to support ${SOME_VARIABLE}. Can this be fixed or is it due to the .yml file extension?

jan-kiszka, 2022-01-04:
One of the reason that speaks against injecting essential information from the build env is that kas configurations are supposed to describe what is needed in order to rebuild the same thing. There can be exceptions from that, and that's why we accept, e.g., `env:` in a controlled manner or allow to set refspec to floating branch names, but you should generally be careful with this.

Can't you model your case by providing kas configuration option files that contain the variations of the URL? Or use `KAS_PREMIRRORS`?

coldspark29, 2022-01-04:
I have no experience with PREMIRRORS. Will have to read about it tomorrow. 

Maybe it isn't necessary, because it is about one package only, which is still on Gerrit. We want to migrate it to our self-hosted Gitlab anyway and then the issue might be resolved for my case.

henning-schild, 2022-01-04:
PREMIRRORS are likely what you want, they can be applied on any layer-cake to steer it to your own mirrors in case you happen to mirror stuff for whatever reason.

It is basically a regexp to be applied on any SRC_URI ... probably much like your `sed` but in a `bitbake` way.

coldspark29, 2022-01-05:
Can you give me an example for my case with the environment variable? I read https://stackoverflow.com/questions/52940431/using-premirrors-in-bitbake-configuration, but I don't really get it.

jan-kiszka, 2022-01-05:
https://kas.readthedocs.io/en/latest/command-line.html comes with an example.

jan-kiszka, 2022-01-05:
And if you want to make that switch options, stick it into a kas option file, analogously to https://github.com/siemens/meta-iot2050/blob/master/kas/opt/mirror-example.yml.

coldspark29, 2022-01-05:
Thanks, but I don't really see how this is going to solve my problem. Guess I don't understand it, but I have migrated the Gerrit repo to Gitlab by now and that has solved my problem.

closed 2022-01-05 by coldspark29