siemens/kas #77: Support for running a command after a repository is cloned

TommyDuc, 2022-02-10:
Hi,

I am currently tasked with porting our various Yocto projects from the `repo` tool to `kas`.
Precisely the project I am trying to port to kas is using the [Wind River setup tool](https://docs.windriver.com/bundle/Wind_River_Linux_Platform_Developers_Guide_LTS_21/page/uqv1630364137783.html).

The workflow we have is basically this:
- Create a project directory and `cd` into it
- Clone the Wind River repository
- Run `setup.sh` with the proper "features" and "templates". Internally this creates a custom `default.xml` file for `repo` and, among other things, will fetch the proper files at the parent project directory level.
- We then clone our layer inside the project directory which is now a Yocto project.
- We run a few other scripts from our layer
- We then invoke `bitbake`

I guess it would be possible to put most of the steps inside a build script, but that would be nicer to have most of this stuff inside the yaml and simply call `kas build`, which our goal to homogenize our builds.

I am a bit new to the Yocto world so if I'm saying nonsense then do not hesitate to point it out :)

TommyDuc, 2022-02-10:
Looking at the other issues it seems https://github.com/siemens/kas/issues/68 is somehow similar

TommyDuc, 2022-02-10:
Hi Jan, thank you for answering quickly!

I think the main problem is one of the egg and the chicken.
Before running the setup tool from WR all there is is a _layer_, not a Yocto project where other layers can be cloned inside.

I understand that the goal of kas is to have everything in terms of bitbake and therefore it might not be the right place to solve this problem but rather on the WR side of things (which is unlikely but who knows maybe I'll get lucky). If that's the case then simply close this issue.

I'll ask your mailing list to see if there are other people who had similar problems.

Thank you for the time.

henning-schild, 2022-02-11:
There are only very little cases where a wrapper around "kas-container" might still be needed, but some times i find myself writing such a wrapper to do additional stuff. But that is mostly only to write down the argv and give "make" experience.
Maybe fetch kas-container, make it executable and verify its sha.

I have code for image compression, image upload to S3 ... stuff that could come after "build". All written in bitbake, which makes it nice and parallel. 
Putting your "post" code into bitbake might be a very good idea for many reasons anyhow.

Same might be true for all the pre stuff, or most of it. Just try to write it in bitbake, a tool you have anyways. And a tool that can do python and shell. So maybe you just move code into a magic entrypoint recipe that everything depends on.
But do not call bitbake inside bitbake.

Also note that a kas yml can point to a repo and include kas bits from there. kas files do not have to live in "layers", it just makes sense to have them there ... often.

`bla.yml`:
```
header:
  version: 10
  includes:
    - repo: meta-iot2050
      file: kas-iot2050-example.yml

build_system: isar

repos:
  meta-iot2050:
    url: https://github.com/siemens/meta-iot2050.git
    refspec: 8d78fb947a32f1b743a91b05844591b9a2580543
```

`kas-container bla.yml`

henning-schild, 2022-02-11:
You can try that last thing right now ... with bla.yml placed in /tmp/bla ... not even a repo ... far from a layer.

jan-kiszka, 2022-08-26:
Issue resolved, some viable solution found?

TommyDuc, 2022-09-04:
> Issue resolved, some viable solution found?

Unfortunately, no, we just did not do the upgrade on this project in the end.

The problem is intrinsic to how `kas`and Wind River's custom build system works and there's not much we can do about it.

closed 2022-09-04 by TommyDuc