siemens/kas #78: Kas is showing errors when updating repos

jclsn, 2022-02-25:
I always get errors when repos are updated. Everything is fine though

```
2022-02-25 11:48:19 - ERROR    - From https://git.openembedded.org/meta-openembedded
2022-02-25 11:48:19 - ERROR    - 23598caea..24f4e6e8d  hardknott-next -> origin/hardknott-next
2022-02-25 11:48:19 - ERROR    - c05ae80ba..0fb490a08  honister       -> origin/honister
2022-02-25 11:48:19 - ERROR    - c05ae80ba..0fb490a08  honister-next  -> origin/honister-next
2022-02-25 11:48:19 - ERROR    - ea223bd3d..cf0ed4239  master         -> origin/master
2022-02-25 11:48:19 - ERROR    - + 67a38f8ed...293e679e7 master-next    -> origin/master-next  (forced update)
2022-02-25 11:48:19 - INFO     - Repository meta-openembedded updated
```

Version is 2.3.3 on Ubuntu Impish. Maybe the version is outdated?

jclsn, 2022-02-25:
Just tried 2.6.3 and same issue

jan-kiszka, 2022-02-25:
That `ERROR` is an annoying side effect of git dumping those messages to stderr. kas just picks up the output streams and classifies them according to stdout vs. stderr. Now, we could say git's stderr is actually stdout, but that may also incorrectly classify real errors. Maybe someone should try to discuss this git upstream?

jclsn, 2022-02-25:
I wouldn't know how to even describe the issue. I still don't understand why there are errors thrown in the first place, when everything works.

jan-kiszka, 2022-02-25:
You can observe that by doing `git fetch 2>git.stderr >git.stdout` against a remote repository that has pending updates since you last updated. Then look at what ends up in the different files. The question for git upstream would be why those update information lines need to go to stderr, rather than stdout. In contrast, when you try that while the remote is offline, the real error will also go to stderr - as it should.

henning-schild, 2022-02-25:
What about `git fetch -q` per default? I guess progress will not be reported on stderr but errors hopefully still.

henning-schild, 2022-02-25:
I will give that a try, stay tuned for a patch.

henning-schild, 2022-02-25:
@jclsn thanks for the report it looks like the problem will be solved soonish

I took the liberty to dig out your email and take things to the list, hope my guess was right.

Please respect our wish for not using issues and place your next issue on the list. When i see your name again here, no matter how urgent/valid the report will be, i will likely not be in the mood to fix it.

jclsn, 2022-02-26:
Okay, thank you for fixing it. I don’t understand your choice regarding mailing lists for issues, but I will use them next time.

commit ce683f2a (2022-02-25): repos: git fetch always with quiet flag

git-fetch prints its progress on stderr which makes harmless progress
output be rendered like ERRORs. Messages we likely do not care about and
certainly do not want to see as errors.
This gets us consistent with most of the other git operations using -q.

Closes: #78
commit 16a69773 (2022-02-26): repos: git fetch always with quiet flag

git-fetch prints its progress on stderr which makes harmless progress
output be rendered like ERRORs. Messages we likely do not care about and
certainly do not want to see as errors.
This gets us consistent with most of the other git operations using -q.

Closes: #78

[Jan: style fix]
closed 2022-03-03 by jan-kiszka