siemens/kas #79: kas-container in pypi release?

priv-kweihmann, 2022-03-23:
Would it be possible to bundle kas-container into the pypi release? - that would would immensely simply the way kas can be setup for some of my CI setups.

I could come up with the needed patches, if that is something the project would support

jan-kiszka, 2022-03-23:
That was not done so far because kas-container does not need the rest of the pip package. It is stand-alone, and you can simply fetch it via fetching e.g. https://raw.githubusercontent.com/siemens/kas/3.0.2/kas-container + a `chmod a+x`.

priv-kweihmann, 2022-03-23:
Okay, I get the motivation - From my perspective just fetching any arbitrary code from a remote host and executing that, still sounds risky - at least if there isn't a sha-checksum available to verify the code just fetched - this would be the case for pypi sdist/wheels.

jan-kiszka, 2022-03-23:
Providing an official sha-sum for the script is a valid point, right. By now, we more and more resolve that by carrying the script in the desired version in the repo that wants to use it (see eg. https://github.com/siemens/meta-iot2050 or https://github.com/siemens/jailhouse-images). Still, if there are other cases, we can consider other solutions. Using pypi for that still looks like a bit of misuse to me. OTOH, it would be technically trivial to add the script to the package. Undecided. Maybe help the discussion with submitting an according patch to the list?

priv-kweihmann, 2022-03-23:
Sure - I'll give it a spin and we could continue the discussion on the mailing list

priv-kweihmann, 2022-04-27:
Patch is in master - closing this issue here

closed 2022-04-27 by priv-kweihmann