siemens/kas #83: sstate_cache support

yli147, 2022-08-17:
Hi,

I'm using 
KAS_CONTAINER_IMAGE=ghcr.io/siemens/kas/kas-isar:3.1 and ./kas-container to cross compile my image. But there is no sstate_cache folder generated.  Even I explicted declare the SSTATE_DIR marco , there is still no shared cache generated. Any idea ?

jan-kiszka, 2022-08-17:
Unless you tweak `KAS_WORK_DIR`, you will find `build/sstate-cache` in the current workdir, ottherwise under `${KAS_WORK_DIR}/sstate-cache`. `SSTATE_DIR` inside the build container is managed by kas as needed.

yli147, 2022-08-17:
No. only below files/folders under build folder
bitbake-cookerdaemon.log  conf  downloads  tmp

yli147, 2022-08-17:
I'm using this project https://github.com/siemens/isar-riscv

jan-kiszka, 2022-08-17:
That project is using an Isar revision pre-dating sstate support.

yli147, 2022-08-17:
Then, how to upgrade the isar to support the sstate, thanks !

jan-kiszka, 2022-08-17:
Raise a PR with isar-riscv, bumping the Isar revision in the related kas config file and likely also updating the recipes to make that new version work. But it's not a kas issue.

yli147, 2022-08-17:
Thanks for the information, will try.

jan-kiszka, 2022-08-26:
Closing as not being a kas topic.

closed 2022-08-26 by jan-kiszka