siemens/kas #85: Error handling in shell script

tewarid, 2022-08-21:
I want to run some commands after a successful kas build, hence my script looks like

```bash
kas build ...
if [ $? -eq 0 ]; then
...
```

Even though kas build fails, the commands still run.

Am I missing something?

jan-kiszka, 2022-08-22:
I cannot confirm this, e.g. by calling `kas build config.yaml --target invalid`. `$?` is non-zero in that case. How exactly did kas fail?

tewarid, 2022-09-23:
It isn't kas itself that fails, but bitbake. kas does print `ERROR    - Command returned non-zero exit status 1` but does not propagate the failure.

tewarid, 2022-09-23:
kas (I'm using version 3.1 now) is propagating bitbake error as expected. Closing issue as I don't see it now.

closed 2022-09-23 by tewarid