siemens/kas #86: Regarding license

orzen, 2022-09-08:
Hi,

It looks like you are using an MIT license. If that is correct, could you please enter the license information for the Github project so that it is available from the project page?

orzen, 2022-09-15:
I think that you should be able to configure the license information in the project settings and select a license explicit. The attach screenshot is what I am referring to, a minor thing but helps out to determine the license for people that are less familiar with open source licenses. 

https://imgur.com/a/i8kWrxP

jan-kiszka, 2022-09-19:
github does not recognize the license file as MIT (in contrast to GPL license files), but it does link to it already. I don't think I can do anything there.

DavidBuechi, 2022-09-19:
GitHub uses [Licensee](https://github.com/licensee/licensee) to detect the license. For the kas project it reports: `Closest non-matching licenses: MIT similarity: 86.51%`
The difference is probably due to the title and the copyright line. GitHubs recommendation:

> To have your license detected, simplify your LICENSE file and note the complexity somewhere else, such as your repository's README file.

But after all it's just a cosmetical "problem" :smirk:

jan-kiszka, 2022-09-19:
Feel free to send a patch against the LICENSE file that adds "MIT License" as header - I bet that does the trick while looking at https://github.com/licensee/licensee/blob/master/LICENSE.md :wink:.

jmillscnhind, 2023-09-13:
@jan-kiszka  Can we pretty please merge this in ? backlash (I know its dumb) is slow adoption.

jan-kiszka, 2023-09-14:
This change has never been proposed to the mailing list or as PR.

commit 30404323 (2024-03-04): make license detectable by licensee

The licensee tool is used by github (and others) to detect the name,
version and attribution of a repositories license. In case of github,
this data is then shown in the summary page, making it easily available
to users. As kas used a non standard header of the license, the tooling
was not able to detect the license with a sufficiently high confidence
to show this as MIT.

This patch makes the header standard and by that reaches a confidence
level of 100% for MIT in the licensee tool.

This patch does NOT change the license itself in any form. It is just
ab [... cut]
commit 1a43ce9d (2024-03-04): make license detectable by licensee

The licensee tool is used by github (and others) to detect the name,
version and attribution of a repositories license. In case of github,
this data is then shown in the summary page, making it easily available
to users. As kas used a non standard header of the license, the tooling
was not able to detect the license with a sufficiently high confidence
to show this as MIT.

This patch makes the header standard and by that reaches a confidence
level of 100% for MIT in the licensee tool.

This patch does NOT change the license itself in any form. It is just
ab [... cut]
commit b40006a3 (2024-03-04): make license detectable by licensee

The licensee tool is used by github (and others) to detect the name,
version and attribution of a repositories license. In case of github,
this data is then shown in the summary page, making it easily available
to users. As kas used a non standard header of the license, the tooling
was not able to detect the license with a sufficiently high confidence
to show this as MIT.

This patch makes the header standard and by that reaches a confidence
level of 100% for MIT in the licensee tool.

This patch does NOT change the license itself in any form. It is just
ab [... cut]
closed 2024-03-07 by jan-kiszka