siemens/kas #88: .bashrc in shell command

lfdmn, 2022-11-16:
I'm new to kas and trying things out. Nice work :) 

When going to  ```kas shell``` is there a way to get ```.bashrc``` sourced?

My usecase is that I define a bunch of aliases I'd like to be able to use when working on recipes.

I tried ```-E, --preserve-env``` argument but it didn't help.

henning-schild, 2022-11-16:
kas and its companion container aim at creating a fresh and pretty isolated environment so that host contamination does not happen, that means you will find your build somewhat isolated from the happy env you know and that is by design

Say you use the container, i imagine you could volume mount some shell script into the container and source that manually. It seems the shell is not a login shell so even if you place it in `/etc/profile.d/` it will not magically work.

```
echo "echo HELLO" > /tmp/hack.sh
kas-container --ssh-dir ~/.ssh/ --runtime-args "-v /tmp/hack.sh:/etc/profile.d/hack.sh:ro" shell kas.yml
source /etc/profile.d/hack.sh
```

henning-schild, 2022-11-16:
if we are talking about a "bash alias" `man 1 bash_builtins` i am afraid that has nothing to do with env.

lfdmn, 2022-11-18:
Hi, 

No there is no container in the loop. I have a LinuxMint 19 distro, pip3 install kass,  made a bare minal config which clones poky and builds core-image-minimal.

When going to the shell the ```ll``` for ```ls - la``` for example isn't defined anymore.

How is the isolation implemented?

lfdmn, 2022-11-21:
OK thanks for the pointers. 

I'll keep investigating kas to see how it could fit my workflows.

closed 2022-11-21 by lfdmn