siemens/kas #91: Add symbolic link feature under repos

syedelec, 2023-02-07:
Hello

I am playing around and I wanted to know if it's possible to integrate the symbolic link feature to reproduce what I was achieving using repo tool

In repo tool, I was doing the following:

```xml
    <project name="meta-custom" remote="remote-name" path="sources/meta-custom" revision="kirkstone">
        <linkfile src="test/script" dest="script"/>  <!--File symlink-->
        <linkfile src="unittests/" dest="unittests"/> <!--Folder symlink-->
    </project>
```

Thanks!

jan-kiszka, 2023-02-07:
Kas does not have the same type of feature. But it does allow to patch repo. Therefore, you could create a patch adding that symlink on the repo itself and include that in the kas config file.

jan-kiszka, 2024-03-08:
Feel free to reopen if the suggestion didn't work.

closed 2024-03-08 by jan-kiszka