siemens/kas #92: feature request : Kas new plugin to set rolling refspec (tag/branches) in yaml manifest to git commits

embetrix, 2023-02-20:
would be nice to have a new plugin to set refspec based on tag/branches to fix git commits 

$ kas fix-ref kas-project.yml > kas-project-fixed.yml

example of kas-project.yml 
```
header:
  version: 11
machine: qemux86
distro: poky
repos:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    refspec: kirkstone
    path: "layers"
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:
  meta-openembedded:
    url: "https://github.com/openembedded/meta-openembedded.git"
    refspec: kirkstone
    path: "layers/meta-openembedded"
    layers:
      meta-oe:
      meta-python:
```

will be converted to kas-project-fixed.yml
```
header:
  version: 11
machine: qemux86
distro: poky
repos:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    refspec: 1cb13d24b592986a44fd653e083b7711513fc916
    path: "layers"
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:
  meta-openembedded:
    url: "https://github.com/openembedded/meta-openembedded.git"
    refspec: bd2d8fba766f912bd1bc5cde7d6f0e72152f1547
    path: "layers/meta-openembedded"
    layers:
      meta-oe:
      meta-python:
```

jan-kiszka, 2023-02-20:
Sound like https://groups.google.com/d/msgid/kas-devel/20230124033239.69433-1-felix.moessbauer%40siemens.com. Please follow up there if you see gaps. I'd like to settle over some solution soon, ideally one that serves most use cases (all is probably not realistic).

embetrix, 2023-02-20:
@jan-kiszka thanks a lot it's exactly what I was looking for !

`kas dump --resolve-refs kas-project.yml > kas-project-fixed.yml
`
shame that the doc doesn't reference it

jan-kiszka, 2023-02-20:
So don't even need that lock-file feature? Also good to know.

Happy to see patches to improve the docs!

jan-kiszka, 2023-02-20:
Indeed, something about `--resolve-refs` is missing there: https://kas.readthedocs.io/en/latest/userguide.html#module-kas.plugins.dump @fmoessbauer

closed 2023-02-20 by embetrix

fmoessbauer, 2023-02-21:
> Indeed, something about --resolve-refs is missing

Yes, I thought I added it but apparently only to the help message. I will add it and also put the examples into code blocks.