siemens/kas #94: kas does not use netrc file

minhix, 2023-02-27:
As the title says, somehow kas does not pick up the netrc.

Info about system:
- kas 3.2.2 (installed by pip)
- Ubuntu 22.10
- Repos on gitlab with 2FA activated. Access tokens are generated and added to a netrc. git clone manually (not via kas) is fine

Command: `NETRC_FILE=./netrc kas shell my_kas_config.yaml` always asks for Username
```
2023-02-27 09:39:50 - INFO     - Repository meta-clang updated
Username for 'https://gitlab.com':
2023-02-27 09:39:51 - INFO     - Repository poky updated
2023-02-27 09:39:55 - INFO     - Repository meta-freescale-3rdparty updated
```

jan-kiszka, 2023-02-27:
Try with an absolute path for NETRC_FILE, that should work. By the time that var is used, the current dir might be different.

jan-kiszka, 2024-03-08:
Feel free to reopen if the issue was not solved meanwhile.

closed 2024-03-08 by jan-kiszka