siemens/kas #95: KAS overwrite PATH environment variable when entering Yocto envrionment

natlor, 2023-02-27:
Hi, I noticed [libkas.py](https://github.com/siemens/kas/blob/0e61fdddaff2dd3f90aabf6ea14212c58d1cae8a/kas/libkas.py#L242_) is overwriting `PATH` environment when calling `source oe-init-build-env/isar-init-build-env`. Is there any reason to change `PATH` instead of using the user `PATH`? 

 I am using an `HOSTTOOLS` installed by `pip` in `/usr/local/bin` but Yocto cannot resolve the `HOSTTOOLS` location as `PATH` doesn't contain anymore `/usr/local/bin`. So, I had to patch `libkas.py` to give `env=None` to `run_cmd()`.

jan-kiszka, 2023-02-27:
Reproducibility of the setup: the goal of kas is that everyone with the config in hands, the kas tool installed and either all official deps installed or using the kas container is able to build things as well.

There is currently a proposal under discussion to address the buildtools use case (https://groups.google.com/d/msgid/kas-devel/20230220122607.1504653-1-alexandre.belloni%40bootlin.com).

jan-kiszka, 2024-03-08:
That feature was never completed, and this request had no follow-up. Feel free to reopen or follow up on the mailing list if there is still a need.

closed 2024-03-08 by jan-kiszka

baum-primes, 2024-10-01:
@jan-kiszka I have exactly the same problem as @natlor: The layer [meta-spdxscanner](https://git.yoctoproject.org/meta-spdxscanner/about/) adds tools to HOSTTOOLS that are not available as system packages, but installed in /usr/local/bin in my case (but I could put them anywhere).
I'd like to discuss possible solutions. How do you propose to continue? Reopen this request (I don't have rights to do so), add a new one, or discuss on the mailing list?

jan-kiszka, 2024-10-02:
It is probably best to start a new discussion as we should focus on how to enable a manageable and reproducible support for `HOSTTOOLS`. Discussions usually take place on the mailing list, but issues can help to pin a topic.