siemens/kas #96: Confusing error when KAS applies patches with quilt

lejcik, 2023-04-04:
Hi all,

I've got a confusing error when I tried to start KAS on my build system:

```
2023-04-03 15:05:45 - ERROR    - /src/myapp/myapp-build/patches/upstream-patches/meta-layer/quilt/
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/kas/kas.py", line 165, in main
    sys.exit(kas(sys.argv[1:]))
  File "/usr/local/lib/python3.8/dist-packages/kas/kas.py", line 154, in kas
    plugin.run(args)
  File "/usr/local/lib/python3.8/dist-packages/kas/plugins/shell.py", line 100, in run
    macro.run(ctx, args.skip)
  File "/usr/local/lib/python3.8/dist-packages/kas/libcmds.py", line 99, in run
    command.execute(ctx)
  File "/usr/local/lib/python3.8/dist-packages/kas/libcmds.py", line 324, in execute
    repos_apply_patches(ctx.config.get_repos())
  File "/usr/local/lib/python3.8/dist-packages/kas/libkas.py", line 196, in repos_apply_patches
    if task.result():
  File "/usr/local/lib/python3.8/dist-packages/kas/repos.py", line 326, in apply_patches_async
    raise FileNotFoundError(p)
FileNotFoundError: /src/myapp/myapp-build/patches/upstream-patches/meta-layer/quilt/
```

It took me a while to find what does this error mean. `FileNotFoundError` is confusing, also the path `.../quilt/` exists. The problem was with `.../quilt/series` file, as by mistake we've added an empty line to it. Its content is like here:

```
0001-my.patch\n
\n
```

KAS joins the `.../quilt/` path with the patch file names that it reads from the `series` file. However, in the case the line is empty, it joins path with empty string, and tries to open it as a file, what fails on `FileNotFoundError`.

I'd suggest to add a condition that will ignore the empty lines, similarly as it does with `#` comments in `repos.py:320`:
`if line.startswith('#'):`

We are on the latest version: `kas 3.2.3 (configuration format version 13, earliest compatible version 1)`

BR,
Vilo

henning-schild, 2023-04-04:
I think i once wrote that code in question. Good catch! How about you send a patch to the mailinglist for that?

In the end it should behave like quilt does. So skip that empty line in case quilt would also skip. Or maybe stop there, or throw some sort of special error.

We also have test-cases which could be extended to have a series file with such an empty line, and assertions that we do "the right thing" as quilt users would expect.

lejcik, 2023-04-05:
Thank you for quick reaction.

Ok, I'll prepare a patch which will fix this issue, also I'll cover such case in the tests. I think it is nothing difficult also for a non-python developer 🙂

I'll be back in few days, I need to find some spare time to do that. As suggested, I'll send the patch to the mailing list...

henning-schild, 2023-04-05:
Thanks for taking care! Feel free to cc me "henning.schild@siemens.com" when you send your patches to the list.

And thanks for the report already, it seems like a pitfall worth fixing.