siemens/kas #101: KAS checkout not working in vscode devcontainer

mschnelte, 2023-09-05:
kas checkout is not working in vscode devcontainers because the needed environment variable 

REMOTE_CONTAINERS_IPC

is not available to the git process.

The checkout fails with "Dev Containers CLI: RPC pipe not configured."

It seems that the env section of the yml file is not parsed during checkout.
Hence adding REMOTE_CONTAINERS_IPC to this section is not sufficient.

Changing the kas/context.py line 109 to:
```        
for key in ['http_proxy', 'https_proxy', 'ftp_proxy', 'no_proxy',
                    'SSH_AUTH_SOCK',
                    'BB_NUMBER_THREADS', 'PARALLEL_MAKE', "q", "REMOTE_CONTAINERS_IPC"]:
```

solved that issue for me.
However would it not be better to parse the env section of the yml file already before checkout?

jan-kiszka, 2023-09-06:
The `env` variables are targeting the bitbake environment, that's why they do not help here. Also, we can't fully parse the `env` key before all repos have been checked out and all kas includes have been merged.

Your case (which no one requested so far) is more comparable to `GIT_CREDENTIAL_HELPER` & Co. Try to extend kas along that line.

jan-kiszka, 2024-03-08:
I suppose it still needs someone to dive deeper into this. I'm not the right candidate as I'm not using vscode. Maybe there are more chances when taking this to the mailing list.

okaestne, 2024-04-08:
I just had the same problem. A few things are necessary for this to work:

`REMOTE_CONTAINERS_IPC` needs to be passed into both, the kas env and to bitbake via `BB_ENV_EXTRAWHITE`. This can be easily be done in the kas.yaml:

```yaml
env:
  # for devcontainer
  REMOTE_CONTAINERS_IPC: null
```

The second part is bitbake's fetch2 cleaned env for git, which needs to contain `REMOTE_CONTAINERS_IPC` as well.

This is only in master right now: https://git.yoctoproject.org/poky/commit/bitbake/lib/bb/fetch2/__init__.py?id=6cb490713ed

So right now, we need to patch bitbake or wait for the next release.

fmoessbauer, 2024-04-14:
@okaestne We have a RFC for that patch on the kas ML for some time now. It would be great if someone could review and test it (ideally with the bitbake from master): https://groups.google.com/g/kas-devel/c/_eQsWitKflo/m/Q3ZhBkr7AQAJ

okaestne, 2024-09-24:
I've just tested this patch on kas 4.5 and scarthgap. In my vscode devcontainer it copies the `.gitconfig` as desired and lets me clone private repos just fine.

fmoessbauer, 2024-09-25:
Ehm... The patch is not applied yet. I'm currently rewriting it to rebase and to add the missing documentation. Please keep this issue open until applied.

okaestne, 2024-09-25:
@fmoessbauer You mention codespaces in your commit. I tested with one of these today as well and found out that they provide the git credentials using a dead-simple git credential helper, that just echoes the parameters, including the `GITHUB_TOKEN` env var as password. So kas would need to pass that one through. This requires a patch for bitbake as well.

I would try to send a patch for bitbake first and a separate one for kas if it's been accepted, or what do you think?

fmoessbauer, 2024-09-26:
> @fmoessbauer You mention codespaces in your commit.

Well... I thought that this works the same way - but apparently I doesn't. I'll just remove this part of the commit message.

> I would try to send a patch for bitbake first and a separate one for kas if it's been accepted, or what do you think?

That sounds like a plan. Anyways, thanks for testing!

commit f5376696 (2024-09-26): add support for running inside vscode devcontainer

When running inside a vscode devcontainer, git credential settings are
injected using a modified .gitconfig file. In addition, the injected
credential handler needs access to environment variables like
REMOTE_CONTAINERS_IPC.

As all this is rather undocumented territory, we first detect if we are
running inside a remotecontainers environment and then forward all
REMOTE_CONTAINERS_* variables.

Xref: https://patchwork.yoctoproject.org/project/bitbake/patch/20231002064106.16886-1-develop@schnelte.de/

Closes: #101
Proposed-by: Matthias Schnelte [... cut]
commit 8773251a (2024-09-30): add support for running inside vscode devcontainer

When running inside a vscode devcontainer, git credential settings are
injected using a modified .gitconfig file. In addition, the injected
credential handler needs access to environment variables like
REMOTE_CONTAINERS_IPC.

As all this is rather undocumented territory, we first detect if we are
running inside a remotecontainers environment and then forward all
REMOTE_CONTAINERS_* variables.

Xref: https://patchwork.yoctoproject.org/project/bitbake/patch/20231002064106.16886-1-develop@schnelte.de/

Closes: #101
Proposed-by: Matthias Schnelte [... cut]
closed 2024-10-01 by jan-kiszka