siemens/kas #102: [Question] Best practices with choices and help in kas menu

jmillscnhind, 2023-09-05:
First off awesome project.  sorry if this is in the wrong place 
Anyways.... 

Ive been playing around with the menu system.  When making choices in the past I do the following.

#### Question 1 choice via  menu

```make
config HAVE_FOO
    bool

config HAVE_BAR
    bool

choice
    prompt  "foo or bar ?"
    default HAVE_FOO
    depends on HAVE_FOO || HAVE_BAR
config TARGET_FOO
    bool "select foo options"
    depends on HAVE_FOO 
 
 # add bar 
 endchoice
 
 KAS_INCLUDE_FOO
 	string
	depends on HAVE_FOO
	default "kas/foo.yml"
```

It seems that I can not use the inter _depends on_  on the _choice_. so the alternative I use _select_

```make
    config TARGET_FOO
        bool "select foo options"
        select HAVE_FOO 
```
 
Is there a better way or a example of how to use _choice_ in the Kconfig for kas? 

#### Question 2 help in the menu.

```make
    config TARGET_FOO
        bool "select foo"
        default y
        help
           Say Y here if you would like too .... 
```

I seen this[1] but can not seem to invoke(? or button) the help section.  Any advice would be great. 

[1] https://github.com/siemens/kas/blob/master/kas/plugins/menu.py#L454

jan-kiszka, 2023-09-06:
We should be compatible with https://www.kernel.org/doc/html/latest/kbuild/kconfig-language.html as we are using a library that aims at this. And that means, you can make a choice element depend on things but not the complete choice group.

Regarding help: This generally works, check existing examples like https://github.com/siemens/meta-iot2050/blob/master/Kconfig. The example excerpt you shared should work as well, though.

JosephMillsAtWork, 2023-09-06:
> We should be compatible with https://www.kernel.org/doc/html/latest/kbuild/kconfig-language.html as we are using a library that aims at this. And that means, you can make a choice element depend on [...]

Jan you're awesome thanks for helping out.  Feel free to close this.

closed 2023-09-08 by jmillscnhind