siemens/kas #103: missing config.yaml after upgrading kas 3.1 -> 3.3

HerrMuellerluedenscheid, 2023-10-04:
Hi,

I upgarded kas from 3.1 to 3.3 and when trying to start the shell I get the following error:

```
kk@bitbake-worker:~/qs-kas$ BB_NUMBER_THREADS=8 kas shell
2023-10-04 15:20:52 - INFO     - kas 3.3 started
2023-10-04 15:20:52 - INFO     - /home/mk/qs-kas$ git rev-parse --show-toplevel
2023-10-04 15:20:52 - INFO     - /home/mk/qs-kas$ git rev-parse --show-toplevel
2023-10-04 15:20:52 - ERROR    - Configuration file not found: /home/mk/qs-kas/.config.yaml
```

Can you tell me how to generate the missing .config.yaml?

Cheers

HerrMuellerluedenscheid, 2023-10-04:
Downgrading to 3.1 I now get the same error. Strange. Maybe I deleted the file by accident? 🤷 
Still, I seem to be in need of a new one...

jan-kiszka, 2023-10-04:
.config.yaml is normally generated by `kas menu`. What changed between 3.1 and 3.3 - among other things - was: https://kas.readthedocs.io/en/latest/format-changelog.html#id12, `_source_dir` and `_source_dir_host`.

You should be able to resolve that issue by re-running `kas menu` over 3.3.

closed 2023-10-06 by HerrMuellerluedenscheid