siemens/kas #104: Published version (4.0) does not include bump to pyyaml dependency, resulting in pip install failures

astapleton, 2023-10-23:
pyyaml fails with cython version 3.0 (see issue: https://github.com/yaml/pyyaml/issues/601). pyyaml 6.0.1 addresses this by pinning the cython build dependency. It seems the dependency restriction to pyyaml <6 was addressed in commit 28007d4, but this change has not been released, meaning that `kas` installations break in certain environments (specifically, this has happened for me when building docker images including kas based on ubuntu 22.04).

Would really appreciate a release to address this breakage!

jan-kiszka, 2023-10-24:
4.1 is on the way, see also https://groups.google.com/d/msgid/kas-devel/adbae121-808a-444f-ab3d-d6583e892596%40siemens.com

Jasper-Ben, 2023-11-22:
I think this issue can be closed now, since we have a 4.1 release?

astapleton, 2023-12-05:
Yep. Thanks!

closed 2023-12-05 by astapleton