siemens/kas #107: Support of repo checkouts created by GIT worktree command

steerpike68, 2023-11-30:
When I checkout a KAS/Yocto based GIT repo branch into a working folder using the GIT worktree feature, the build fails at an early stage:

$ kas/kas-container -v --ssh-dir ${HOME}/.ssh  build kas/project-latest.yml
+ docker run -v /<worktreepath>:/repo:ro -v /<worktreepath>:/work:rw -e KAS_WORK_DIR=/work -v /<worktreepath>/build:/build:rw --workdir=/repo -e KAS_BUILD_DIR=/build -e USER_ID=1000 -e GROUP_ID=1000 --rm --init -v /${HOME}/.ssh:/var/kas/userdata/.ssh:ro -t -i -e TERM=xterm-256color -e SHELL=/bin/bash --log-driver=none --user=root ghcr.io/siemens/kas/kas:4.0 -d build /repo/kas/project-latest.yml
2023-11-30 10:40:40 - INFO     - kas 4.0 started
2023-11-30 10:40:40 - DEBUG    - Using selector: EpollSelector
2023-11-30 10:40:40 - INFO     - /repo/kas$ git rev-parse --show-toplevel
2023-11-30 10:40:40 - INFO     - /repo/kas$ hg root
2023-11-30 10:40:40 - INFO     - /repo/kas$ git rev-parse --show-toplevel
2023-11-30 10:40:40 - INFO     - /repo/kas$ hg root
2023-11-30 10:40:40 - DEBUG    - execute setup_dir
2023-11-30 10:40:40 - DEBUG    - execute setup_home
2023-11-30 10:40:40 - DEBUG    - execute init_setup_repos
2023-11-30 10:40:40 - ERROR    - Configuration file not found: /repo/kas/kas/includes/default-config.yml

I've replaced the local worktree folder by <worktreefolder> in the command. This worktree contains a .git file which is pointing to the main repo this worktree is based upon. I assume that the current kas version instead expects a .git folder.

When I instead create the local folder by cloning the repo and checkout the correct branch, the same command works fine.

steerpike68, 2023-12-01:
OK, I don't think its a problem of the config files or if includes are used.
To reproduce, just take a simple kas-container based project you already have as a git repo with at least 2 different branches which I here call A and B. First clone the repo to folder /some_path/A/.git and check out branch A into A. Then cd to the folder A and start the kas-container build by the command you normally would use to build the project. That should work.
Then from folder A use command "git worktree add ../B B" to create a second checkout folder B of branch B. "git worktree list" should show now:
/some_path/A          c-id [A]
/some_path/B          c-id [B]
Then do "cd ../B" and use the same build command as used for folder/branch A. This should fail - even if that project doesn't use any includes at all.

jan-kiszka, 2023-12-02:
Cannot confirm, even when using a kas file that comes with includes.

commit af56de47 (2025-01-07): kas-container: support git worktrees

Git support sharing the .git directory across multiple worktrees each having
its own HEAD, index, rebase-todo and so on. The worktree directory (created
with `git worktree`) contains no directory .git, but a file .git with a
reference to the main git directory. Therefore, this directory must also be
mounted in the container to make git operable.

Git uses an absolute path to anchor the worktrees to the common
repository. As we do not want to recreate the full host path inside the
container, we mount the common repo as /repo-common and patch the .git
file o [... cut]
closed 2025-01-08 by jan-kiszka