siemens/kas #108: libkas ssh_cleanup_agent() function could be less noisy

JosephMillsAtWork, 2024-01-31:
Example: 

Say I have set the SSH_PRIVATE_KEY or SSH_PRIVATE_KEY_FILE in the env for an CI. In my CI I would like to gather information from bitbake. Or maybe I want to output kas dump to a file. 

```
FOO="$(kas shell /some/yaml.yml -c "bitbake -e | grep ^DISTRO")"
echo $FOO
``` 
expected output 

```
poky
```

received output 

```
poky 
unset SSH_AUTH_SOCK;
unset SSH_AGENT_PID;
echo Agent pid .....
```

```
kas dump --resolv-ref --resolv-env /some/yaml.yml > /tmp/file.txt
```

This will also have the unset parts in it at the top. 

proposed change is to add stdout=None in libkas.py->ssh_cleanup_agent. 

Thanks for your time.

JosephMillsAtWork, 2024-01-31:
Patchfile.

[quiet-ssh-cleanup-agent.txt](https://github.com/siemens/kas/files/14116880/quiet-ssh-cleanup-agent.txt)

JosephMillsAtWork, 2024-02-01:
> Patchfile.
> 
> [quiet-ssh-cleanup-agent.txt](https://github.com/siemens/kas/files/14116880/quiet-ssh-cleanup-agent.txt)

After testing the above patch , sadly it is still not working. Moving over to subprocess.DEVNULL seemed to do the trick. Attached is a patch. 
[quiet-downnow-process.txt](https://github.com/siemens/kas/files/14118850/quiet-downnow-process.txt)

jan-kiszka, 2024-02-01:
Thanks for reporting and working on this! Just please have a look at https://github.com/siemens/kas/blob/master/CONTRIBUTING.md as well in order to get your patch into the final shape in the right channel.

fmoessbauer, 2024-02-12:
@JosephMillsAtWork Please have a look at [this patch](https://groups.google.com/g/kas-devel/c/oX29M3VGtkw/m/MQsNs2xPAQAJ). That should also fix the issue you reported. If you like to get a `Reported-by:`, please just answer on that mail and add the `Reported-by:` line.

closed 2024-03-08 by jan-kiszka