siemens/kas #110: 4.3 introduces some breaking change when using just SSH_PRIVATE_KEY environment variable

Brend-Smits, 2024-03-12:
Hey! 

First of all, thanks for creating KAS. We've been happily using it for some time now!
Unfortunately, it seems like today some kind of bug/breaking change was introduced causing our CI pipelines to no longer function.
In our CI workflow we set the environment variable `SSH_PRIVATE_KEY` to a private key, which should indicate to KAS that the internal SSH Agent should be used with this key. This worked fine up until KAS version 4.2, in version 4.3 it broke.

Here's the log:
```
2024-03-12 08:08:08 - INFO     - kas 4.3 started
2024-03-12 08:08:08 - INFO     - adding SSH key
2024-03-12 08:08:08 - ERROR    - failed to add ssh key: b'Could not open a connection to your authentication agent.\n'
2024-03-12 08:08:08 - INFO     - Cloning repository XX
2024-03-12 08:08:08 - INFO     - Cloning repository YY
2024-03-12 08:08:08 - ERROR    - Command "/__w/yocto-build/yocto-build/workspace$ git clone -q git@github.com:org-here/XX.git /__w/yocto-build/yocto-build/workspace/XX" failed
--- Error summary ---
Warning: Permanently added the RSA host key for IP address 'Z.ZZZ.ZZ.[19](https://github.com/org-here/yocto-build/actions/runs/8240283299/job/22549425967#step:7:20)7' to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
 - ERROR    - fetch repos failed: error code 1[28](https://github.com/org-here/yocto-build/actions/runs/8240283299/job/22549425967#step:7:29)
```

Please let me know if I can help to debug this further or if you need more information.

jan-kiszka, 2024-03-12:
Yeah, there is likely an issue, see also https://groups.google.com/g/kas-devel/c/VwkqOz5F3yc/m/-SFPL93PAQAJ. @fmoessbauer is looking into it.

jan-kiszka, 2024-03-12:
Current version in next should have the issue fixed.

closed 2024-03-13 by jan-kiszka

Brend-Smits, 2024-03-13:
Tested it and it's working again. Thanks a lot for the swift resolution!

commit 5b1558ad (2024-03-13): fix(ssh-setup-agent): correctly parse result

In dbd32b04 we switched to run_cmd to execute the ssh-agent. However,
this wrapper returns the output as a single string instead of
line-chunks. By that, we iterated over the single characters instead of
the lines which broke the key-value parsing.

This patch fixes this by splitting the string at the end-of-line
delimiter.

Reported-by: Jasper Orschulko <Jasper.Orschulko@iris-sensing.com>
Closes #110