siemens/kas #111: .netrc should not contain comments

adamjohnstonarm, 2024-03-25:
A recent change to OE core causes bitbake to fallback to netrc when
BB_HASHSERVE_USERNAME or BB_HASHSERVE_PASSWORD are not set.

This exposes us to an error when running in GitLab because kas has been
inserting a comment into ~/.netrc.

`# appended by kas, you have gitlab CI env`

Resulting in...

`netrc.NetrcParseError: bad toplevel token 'appended'`

The comment should not be added is it relies on undocumented behaviour.

adamjohnstonarm, 2024-03-25:
Fix sent to mailing list

adamjohnstonarm, 2024-03-25:
See #112

jonmason, 2024-04-05:
Can we get this in a tagged release, as we're having to carry a patch in our CI to work around it?  Perhaps 4.3.2.

jan-kiszka, 2024-04-08:
Working towards this, should appear this week.

fmoessbauer, 2024-04-10:
We now have the 4.3.2 release, which includes the fix. By that, we can close this issue. @jan-kiszka

closed 2024-04-10 by jan-kiszka