siemens/kas #113: Running `kas-container` inside docker image `kas:latest`

roby2014, 2024-03-27:
Hi. I'm trying to use `kas-container` inside a Docker image itself (`ghcr.io/siemens/kas/kas:latest`) provided on documentation page.

```
kas-docker-image:
  stage: build
  image: ghcr.io/siemens/kas/kas:latest
  when: manual
  allow_failure: false
  script:
    - ./build.sh -docker $MACHINE
 ``
 
However, the `kas:latest` image doesn't have the docker engine, so this fails.
Any idea how I can achieve this?

jan-kiszka, 2024-03-28:
What exactly are you trying to do? Use kas in CI, and there inside the kas-provided container image? In that case, just call kas, not kas-container. See e.g. https://github.com/siemens/meta-iot2050/blob/master/.gitlab-ci.yml. If you want to use kas-container, there is no need to specify the image. Here is an example for that pattern (though on github): https://github.com/siemens/meta-iot2050/blob/master/.github/workflows/main.yml.

BTW, what is that build.sh? It's usually an anti-pattern to wrap kas in another script.

roby2014, 2024-03-29:
In short, my idea is to run my CI in a docker image (which installs kas, docker, python, pip), and inside that image I use `kas-container` to clean compile each of my board targets. If I have containers, I can later access the build files and check/debug if I need for only the specific board I want.

`build.sh` is just a script that either runs `kas` or `kas-container` depending on `-docker` argument (so I can also use this locally without the need of a container build)

jan-kiszka, 2024-03-29:
As you can see from the first example I shared, there is no reason for that extra wrapping. Our official kas container holds everything you need, just use that directly. And `kas` is just like `kas-container` in most of the cases, so call the former directly inside CI.

roby2014, 2024-03-30:
Im a bit confused then. Whats the difference between kas and kas-container?
Your first example uses your stock image, which doesnt have docker engine installed so kas-container wont work.

jan-kiszka, 2024-03-30:
The first example asks Gitlab CI to run all steps inside the container image provided by kas, thus implicitly asks for a container runtime as well. And as we are inside a kas container then, we no longer need the `kas-container` script which does that very same wrapping when you call it locally.

roby2014, 2024-04-04:
Thank you. Understood. My approach will be to just run `kas-container`, since the only dependency is a docker engine on my runner.