siemens/kas #116: Possible to change KAS_BUILD_DIR within kas yaml config?

rcross-lc, 2024-05-03:
My use case here is that I have a kirkstone.yaml and a scarthgap.yaml and I would like them to have completely separate build directories.

Is there a way to set this up so that I don't need to manage environment variables via another wrapper script?

If not, can this setting be exposed to the yaml config?

jan-kiszka, 2024-05-03:
`KAS_BUILD_DIR` is not really a project setting, it's a local decision. Or why should all users of your configs split the build dirs like you suggest?

But even if we wanted to introduce a configurable default value to the config, that would not work across `kas-container`. It needs to know this value but has no parser for configs with all their includes and overrides (except for very simplistic top-level evaluations).

rcross-lc, 2024-05-06:
I think I understand the problem, I'll investigate why my scarthgap builds can't co-exist with my kirkstone builds and/or change the TMPDIR and cache directories maybe...

nicola-lunghi, 2026-09-29:
Hi @jan-kiszka this is not resolved I think

As I have multiple configurations, I want to specify the work_dir and build_dir for each of them. for example I have a build with musl enabled and one without, and it would be best to have multiple build dir. Also there's a way to share the sstate cache and downloads folder between multiple builds?

thanks
Nicola