siemens/kas #117: Access to private repositories with kas & gitlab CI/CD

redeexpressos, 2024-05-07:
Hello. I'm trying to do a simple CI/CD that runs kas docker image and builds my image (which depends on private repositories)
This is what I've tried:
```yml
variables:
  GIT_SUBMODULE_STRATEGY: recursive

before_script:
  - git config --global credential.helper store
  - echo "https://gitlab-ci-token:${CI_JOB_TOKEN}@my.org.com/git" > ~/.git-credentials
  - git config --global url."https://gitlab-ci-token:${CI_JOB_TOKEN}@my.org.com/git".insteadOf ssh://git@my.org.com:1337

stages:
  - build
  
build:
  stage: build
  image: ghcr.io/siemens/kas/kas:4.3.2
  timeout: 48 hours
  script:
    - kas build
  artifacts:
    paths:
      - build/tmp/deploy/images/*/*tegraflash.tar
```

However, when `kas` tries to clone, it still tries the `ssh` way.
Any suggestion how I can do this?

fmoessbauer, 2024-05-07:
Hi, for this use-case, you basically have three options:

- use a ssh deploy key key and add the filename to `SSH_PRIVATE_KEY_FILE`. By that, you can still clone via SSH (and independent of the access rights of the user that triggers the CI).
- Use your approach from above, but point `GITCONFIG_FILE` to your file. Currently we only auto-use that file in the Github CI (1f6197b3c2ef6687b645178bfdb6b1f9f36f4e03)
- Use `KAS_PREMIRRORS` to modify the urls on the fly.

In the manual, we have a dedicated [credential handling section](https://kas.readthedocs.io/en/latest/userguide/credentials.html#credential-handling), but if things are still not clear we are happy to add further pointers / examples.

redeexpressos, 2024-05-07:
Is there any available example using the kas's option `CI_JOB_TOKEN`?

fmoessbauer, 2024-05-07:
> Is there any available example using the kas's option `CI_JOB_TOKEN`? I am not sure how to use these options..

Unfortunately not, but you're right, we should add one.

The probably easiest solution is to do the following:

```yml
variables:
  # adjust according to your needs
  KAS_PREMIRRORS: "git@code.siemens.com: https://code.siemens.com/"
```

Then, make sure that the `CI_JOB_TOKEN` has the necessary access rights on the "other" project. For that, navigate to the other project in Gitlab -> CI/CD -> Token Access. There, tick "Limit access to this project" and add the name of the project that executes the ci (group/name). Done.

redeexpressos, 2024-05-07:
Thanks, that did the trick :) And yes, I do agree there should be more examples with the various environment variables.

fmoessbauer, 2024-05-07:
I just sent a documentation patch to the kas ML: https://groups.google.com/g/kas-devel/c/ilW8NOQfZV0