siemens/kas #118: kas dump yaml file order

embetrix, 2024-05-23:
kas-example.yml:

```
header:
  version: 11
machine: qemux86-64
distro: poky
target:
  - core-image-minimal
repos:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    branch: scarthgap
    path: "layers"
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:
  meta-openembedded:
    url: "https://github.com/openembedded/meta-openembedded.git"
    branch: scarthgap
    path: "layers/meta-openembedded"
    layers:
      meta-oe:
      meta-python:
      meta-networking:
      meta-perl:
      meta-python:
      meta-multimedia:
      meta-filesystems:
  meta-swupdate:
    url: "https://github.com/sbabic/meta-swupdate.git"
    branch: scarthgap
    path: "layers/meta-swupdate"
env: 
    DL_DIR         : "${TOPDIR}/downloads"
    SSTATE_DIR     : "${TOPDIR}/sstate-cache"
local_conf_header:
  standard: |
    CONF_VERSION = "2"
    INHERIT += "rm_work"
  debug-tweaks: |
    EXTRA_IMAGE_FEATURES = "debug-tweaks"
```

kas dump --resolve-refs kas-example.yml  > kas-example-fixed.yml

kas-example-fixed.yml:

```
distro: poky
env:
    DL_DIR: ${TOPDIR}/downloads
    SSTATE_DIR: ${TOPDIR}/sstate-cache
header:
    version: 11
local_conf_header:
    debug-tweaks: |
        EXTRA_IMAGE_FEATURES = "debug-tweaks"
    standard: |
        CONF_VERSION = "2"
        INHERIT += "rm_work"
machine: qemux86-64
repos:
    meta-openembedded:
        branch: scarthgap
        commit: 6de0ab744341ad61b0661aa28d78dc6767ce0786
        layers:
            meta-filesystems:
            meta-multimedia:
            meta-networking:
            meta-oe:
            meta-perl:
            meta-python:
        path: layers/meta-openembedded
        url: https://github.com/openembedded/meta-openembedded.git
    meta-swupdate:
        branch: scarthgap
        commit: ae2f52ce3c8ee742218715d1d542eaae3b4ca19c
        path: layers/meta-swupdate
        url: https://github.com/sbabic/meta-swupdate.git
    poky:
        branch: scarthgap
        commit: 4b07a5316ed4b858863dfdb7cab63859d46d1810
        layers:
            meta:
            meta-poky:
            meta-yocto-bsp:
        path: layers
        url: https://git.yoctoproject.org/git/poky
target:
- core-image-minimal

```

would be possible to keep the same original structure for kas-example-fixed.yml:

```
header:
  version: 11
machine: qemux86-64
distro: poky
target:
  - core-image-minimal
repos:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    branch: scarthgap
    commit: 4b07a5316ed4b858863dfdb7cab63859d46d1810
    path: "layers"
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:
  meta-openembedded:
    url: "https://github.com/openembedded/meta-openembedded.git"
    branch: scarthgap
    commit: ae2f52ce3c8ee742218715d1d542eaae3b4ca19c
    path: "layers/meta-openembedded"
    layers:
      meta-oe:
      meta-python:
      meta-networking:
      meta-perl:
      meta-python:
      meta-multimedia:
      meta-filesystems:
  meta-swupdate:
    url: "https://github.com/sbabic/meta-swupdate.git"
    branch: scarthgap
    commit: ae2f52ce3c8ee742218715d1d542eaae3b4ca19c
    path: "layers/meta-swupdate"
env: 
    DL_DIR         : "${TOPDIR}/downloads"
    SSTATE_DIR     : "${TOPDIR}/sstate-cache"
local_conf_header:
  standard: |
    CONF_VERSION = "2"
    INHERIT += "rm_work"
  debug-tweaks: |
    EXTRA_IMAGE_FEATURES = "debug-tweaks"

```

fmoessbauer, 2024-05-23:
Hi @embetrix , despite the changed order, both the input and the combined output should be semantically identical. If not, this is a bug. As kas internally represents the configuration as a python dict, the order is currently not preserved.

Do you have a use-case where preserving the order is required (or desired)? Ideally, you could also bring this topic up on the mailing list (kas-devel@googlegroups.com), as we have a broader audience there.

embetrix, 2024-05-23:
@fmoessbauer I agree that the semantic and usage are identical even the order has changed. 
My use case is that I would like that for releases using a fixed dump of the kas yaml file and track the changes easily in Git,
with a changed order it's hard to track.

embetrix, 2024-05-23:
I think that kas dump is sorting the yaml keys alphabetically, I believe it can be turned off.

fmoessbauer, 2024-05-23:
> I think that kas dump is sorting the yaml keys alphabetically, I believe it can be turned off.

By that I expect to get a rather stable ordering. Without sorting, the keys are likely at nondeterministic positions, which makes thinks even more unstable.

At least for json there is a "canonical" format that is stable (keys are sorted, no whitespacing). I don't know if a similar standard also exists for yaml. This however, is likely not the solution you are looking for.

embetrix, 2024-05-23:
maybe as an option parameter for kas dump : do not sort ?

gowufang, 2024-12-11:
If the kas-example.yml at the beginning is configured in dictionary order, then the layout of kas-example-fixed.yml will not be so different. lol

fmoessbauer, 2024-12-11:
@gowufang I re-checked the code and actually found the inconsistency. When doing `yaml.dump()`, keys are automatically sorted, while `json.dump` does not sort the keys. I'll propose a change to only sort the keys if `--sorted` is specified on the command line.

commit df267f41 (2024-12-21): dump: only sort keys if explicitly requested

The yaml and json formatter differed w.r.t sorting the keys during
serialization (json kept the order, yaml sorted). As pointed out in #118
the users expect to maintain the order of the keys. By that, we make
this the new default and add an CLI option "--sort" to sort the output.

Closes: #118
closed 2025-01-03 by jan-kiszka