siemens/kas #120: `ghcr.io/siemens/kas/kas-isar:latest` missing HOSTTOOLS

redeexpressos, 2024-06-03:
Using ` ghcr.io/siemens/kas/kas-isar:latest` and `kas build platform` on my gitlab CI/CD, however, getting this error:
```
2024-06-03 14:57:43 - INFO     - ERROR: The following required tools (as specified by HOSTTOOLS) appear to be unavailable in PATH, please install them in order to proceed:
2024-06-03 14:57:43 - INFO     - chrpath cpio cpp g++ gawk gcc
```

Shouldnt these tools be available by default?

jan-kiszka, 2024-06-03:
I suspect you are trying to use the kas-isar container to build a yocto/OE image, aren't you? That's what ghcr.io/siemens/kas/kas is for. kas-isar supports building Debian images with the isar (https://github.com/ilbers/isar) build system.

redeexpressos, 2024-06-03:
Oooops. Thank you, using `image: ghcr.io/siemens/kas/kas:latest` works.