siemens/kas #122: KAS checkout can't handle fast-forward merges

lms-jk, 2024-06-10:
Our workflow involves fast-forward merges. Switching a meta-layer branch from a merged feature-branch back to the release branch does not work with kas checkout.

As an example, `scarthgap` is the release branch and `devel` is the feature branch of a meta-layer repository `bsp`. During development, the meta-layer was configured to track the feature branch `devel` in the config file:
```yaml
...
repos:
  bsp:
    url: "<our_internal_git_server>"
    branch: devel
    commit: "fab4308300b7b7df5dec4e88bdea2b8c8c91f2cc"
    path: "layers/meta-lms-nvidia-bsp"
...
```
After merging the meta-layer with a fast-forward merge (which does not change the commit hash), the release branch is tracked again:
```yaml
...
repos:
  bsp:
    url: "<our_internal_git_server>"
    branch: scarthgap
    commit: "fab4308300b7b7df5dec4e88bdea2b8c8c91f2cc"
    path: "layers/meta-lms-nvidia-bsp"
...
```
This change will produce the following error message, although "fab4308300b7b7df5dec4e88bdea2b8c8c91f2cc" does exist in branch scarthgap now:
> 2024-04-10 13:56:34 - ERROR - Branch "scarthgap" in repository "bsp" does not contain commit "fab4308300b7b7df5dec4e88bdea2b8c8c91f2cc"

My hypothesis is, that KAS does not fetch the latest revisions from repository `bsp`, because the commit hash already matches. This also means, that KAS misses the update of branch `scarthgap` and complains, that said commit is not contained in the branch.

fmoessbauer, 2024-06-10:
Hi!

> My hypothesis is, that KAS does not fetch the latest revisions from repository bsp, because the commit hash already matches. This also means, that KAS misses the update of branch scarthgap and complains, that said commit is not contained in the branch.

This is likely the reason. We indeed need to check if we have the commit on that branch as well and execute the fetch operation otherwise. I just want to double check if this is a regression in 4.3, or if this has always been like that. Did that work in the 4.1 / 4.2 version?

lms-jk, 2024-06-10:
I will check and come back to you.

lms-jk, 2024-06-11:
First I need to add, that appending `--update` to either `kas checkout` or `kas build` fixes this issue.

Testing for regressions:
- 4.1 and 4.2 on Ubuntu 20.04 do not checkout any branch when the commit is also specified (this functionality was introduced in [4.3](https://github.com/siemens/kas/blob/37f843356b72c91e2833c7e486ff433b4eeb1db2/CHANGELOG.md?plain=1#L35))
- 4.3, 4.3.1 on Ubuntu 20.04 are affected
- 4.3.2 and 4.4 on Ubuntu 22.04 are affected

fmoessbauer, 2024-06-13:
> First I need to add, that appending --update to either kas checkout or kas build fixes this issue.

Yes, that's what I expected. Our short circuit to avoid the remote fetching was simply not precise enough. When adding `--update`, a fetch is always performed.

I just pushed a fix to the ML (you are in CC): https://groups.google.com/g/kas-devel/c/UmRKGPu_hPo. Please give it a try.

lms-jk, 2024-06-13:
Works perfectly. Thank you!

commit 57c2b36c (2024-06-13): fix: correctly handle upstream ff merges on fetch

In case both branch and commit are specified, kas checks if the commit
is contained in the branch. Further, only when a ref was not present in
the local clone, a fetch from remote was executed. This logic however
breaks on upstream fast-forward merges, as these do not create a new
commit id, but instead just move the branch head.

This commit adds a check if we also have the requested commit sha on the
requested branch locally and if not continues to the fetching step
(instead of the short circuit).

Fixes: #122
Reported-by: Jan Kircher <jan.k [... cut]
closed 2024-06-14 by lms-jk