siemens/kas #130: bblayers order when using path

kwavnet, 2024-10-08:
I have the following two example snippets

base.yml

```yaml
header:
  version: 17
distro: poky
repos:
  poky:
    url: "https://git.yoctoproject.org/git/poky"
    tag: kirkstone-4.0.20
    path: sources/poky
    name: aaaa_poky
    layers:
      meta:
```

and layer1.yml

```yaml
header:
  version: 17
repos:
  somethingelse:
    url: "https://github.com/somethingelse/meta-somethingelse"
    branch: master
    path: sources/meta-somethingelse
    name: zzzz_somethingelse
```

I want to explicitly use name to create some pseudo ordering of the layer (sic 🙄 , I know, but sadly there are vendor layers that need to be included before others).

which ends up with

```
BBLAYERS ?= " \
    ${TOPDIR}/../sources/meta-buildutils \
    ${TOPDIR}/../sources/poky/meta"
```

in the `bblayers.conf`.
That feels wrong IMO.
I would have expected a alphabetical ordering by `repo-id` and/or `name` from what I read from the current docs.

BTW tests have been done with kas 4.5.

If I now remove the path attribute I end up with

```
BBLAYERS ?= " \
    ${TOPDIR}/../aaaa_poky/meta \
    ${TOPDIR}/../zzzz_somethingelse"
```

which is the order I would have expected, but now the name of the paths make it extra hard to guess what is checked out.

Is that expected behavior?
Did I miss it in the documentation, or might that be something to explicitly state?

https://github.com/siemens/kas/issues/36#issuecomment-784819120 mentions ordering by name, which is not what I see in my example, hence the question

fmoessbauer, 2024-10-08:
Hi,

this behavior is actually expected (if you read the documentation carefully), but it probably can be improved to be more useful. The layers are simply sorted by (final) path, whereby it does not matter how the path is constructed. If you specify it explicitly, that's it. If you don't specify `path`, then the repo is checked out as `${KAS_WORK_DIR}/repo-name`, whereby `repo-name` is the `repo.name` if set, or `repo-id` otherwise. This is implemented here: https://github.com/siemens/kas/blob/12add6c6ec92554a7e77623a6a07be5899a71460/kas/libcmds.py#L419

By that, you should be able to implement you use-case by simply dropping the "path". Nonetheless, I'll send a patch to change the behavior as following:

1. sort by repo (name/id)
2. sort by layer name (just the name, not the full path)

This also solves the problem with absolute path, whereby the user currently cannot influence the order.

kwavnet, 2024-10-08:
> Hi,
>
> this behavior is actually expected (if you read the documentation carefully), but it probably can be improved to be more useful. The layers are simply sorted by (final) path, whereby it does [...]

@fmoessbauer thanks for the explanation and a potential patch

fmoessbauer, 2024-10-08:
@kwavnet Can you please give https://github.com/fmoessbauer/kas/commit/7b65bb0ca52675e591d2a29e748d8815f42cdf3a a try. If you are using the kas container, just use `ghcr.io/fmoessbauer/kas/kas:next`.

kwavnet, 2024-10-08:
> @kwavnet Can you please give [fmoessbauer@7b65bb0](https://github.com/fmoessbauer/kas/commit/7b65bb0ca52675e591d2a29e748d8815f42cdf3a) a try. If you are using the kas container, just use `ghcr.io/fmoessbauer/kas/kas:next`.

Fantastic response time 🚀 - I can confirm your patch does solve my issue - a huge thank you from my side

commit ecf5827b (2024-10-09): repos: order layers by repo and layer name

Usually the order in which layers are appended to the bblayers does not
matter, but there are cases where fine-grained control is needed.

Before this patch, all layers were sorted by the final path. This
however creates issues when specifying both the path and the name, as
well as when absolute paths are used.

We change this by first sorting by repo name and then layer path. We
further document this behavior in the project configuration.
As the repo name was already part of the full path before, it is very
unlikely that downstream users are affecte [... cut]
closed 2024-10-19 by jan-kiszka